realsenseai/librealsense #12116: the source code of align function with python

puyiwen, 2023-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | Open RealSense Viewer  | 
| Operating System & Version |  Linux  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python }                          | 
| Segment			|  Robot }                          | 

### Issue Description
I would like to know where is the corresponding source code for implementing the align function in the pyrealsense2 library? Thank you very much!!

MartyG-RealSense, 2023-08-21:
Hi @puyiwen  The RealSense SDK has a pyrealsense2 example program for depth to color alignment called **align_depth2color.py** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

puyiwen, 2023-08-21:
> 你好@puyiwen RealSense SDK 有一个用于深度到颜色对齐的pyrealsense2 示例程序，名为**align_depth2color.py，**位于下面的链接。
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

Thank you for your reply, the code you provided has used the **align API**, and I would like to know what is the implementation of the **align API** interface of this module?

closed 2023-09-05 by MartyG-RealSense