realsenseai/librealsense #12119: 在vs2022 C# 使用Realsense d435i 獲取的深度值與Intel.RealSense.Viewer.exe不一樣且不穩定

ned6602, 2023-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d435i | 
| Firmware Version                   | Intel.RealSense.Viewer.exe--> v.2.54.1) | 
| Operating System & Version |                            Win 11                        | 
| Platform                                 |                               PC                            
| SDK Version                           | [image]
| Language                               |                    C# windows form.Net framework 4.8     | 
| Segment			         |                              Robot                       | 

### Issue Description
影片
https://github.com/IntelRealSense/librealsense/assets/134619419/0803cb46-781d-4eb0-8cbb-ef7089a95e7c

我的原始程式碼
https://drive.google.com/file/d/1XKB4ZC8sAG8yXJNgB7Cd7qW7QtTDe_ot/view?usp=sharing

MartyG-RealSense, 2023-08-21:
Hi @ned6602  The RealSense Viewer tool applies a range of post processing filters by default that can help to stabilise depth.  With a program script such as C# there are no post processing filters applied because they have to be deliberately programmed into the script.  An example of defining post-processing filters in C# is at https://github.com/IntelRealSense/librealsense/issues/10272

You can read more about post-processing at the link below.

https://dev.intelrealsense.com/docs/post-processing-filters

**********

您好 @ned6602 RealSense 查看器工具默認應用一系列後處理過濾器，有助於穩定深度。 對於諸如 C# 之類的程序腳本，不會應用後處理過濾器，因為必須將它們特意編程到腳本中。 在 C# 中定義後處理過濾器的示例位於 https://github.com/IntelRealSense/librealsense/issues/10272

您可以在下面的鏈接中閱讀有關後處理的更多信息。

https://dev.intelrealsense.com/docs/post-processing-filters

ned6602, 2023-08-22:
我有看到濾波器順序
Depth Frame >> Decimation Filter >> Depth2Disparity Transform** -> Spatial Filter >> Temporal Filter >> Disparity2Depth Transform** >> Hole Filling Filter >> Filtered Depth.
但是我只有DisparityTransform();沒有Depth2Disparity Transform跟Disparity2Depth Transform
我目前的代碼
`block = new CustomProcessingBlock((f, src) =>
                {
                    using (var releaser = new FramesReleaser())
                    {
                        foreach (ProcessingBlock p in blocks)
                            f = p.Process(f).DisposeWith(releaser);
                        var decimationFilter = new DecimationFilter();
                        var DisparityTransform = new DisparityTransform();
                        var spatialFilter = new SpatialFilter();
                        var temporalFilter = new TemporalFilter();
                        var holeFillingFilter = new HoleFillingFilter();
                        var thresholdFilter = new ThresholdFilter();
                        
                        f = f.ApplyFilter(decimationFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(DisparityTransform).DisposeWith(releaser);
                        f = f.ApplyFilter(spatialFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(temporalFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(holeFillingFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(thresholdFilter).DisposeWith(releaser);

                        f = f.ApplyFilter(align).DisposeWith(releaser);
                        f = f.ApplyFilter(colorizer).DisposeWith(releaser);

                        var frames = f.As<FrameSet>().DisposeWith(releaser);

                        var colorFrame = frames[Stream.Color, Format.Rgb8].DisposeWith(releaser);
                        var colorizedDepth = frames[Stream.Depth, Format.Rgb8].DisposeWith(releaser);

                        var res = src.AllocateCompositeFrame(colorizedDepth, colorFrame).DisposeWith(releaser);
                        src.FrameReady(res);
                    }
                });`
                

影片
                

https://github.com/IntelRealSense/librealsense/assets/134619419/89cd638e-d0b7-453c-89b1-56300e082a60

結果還是不理想

ned6602, 2023-08-22:
我有看到濾波器順序
Depth Frame >> Decimation Filter >> Depth2Disparity Transform** -> Spatial Filter >> Temporal Filter >> Disparity2Depth Transform** >> Hole Filling Filter >> Filtered Depth.
但是我只有DisparityTransform() 沒有Depth2Disparity Transform跟Disparity2Depth Transform
這是我目前的代碼
`block = new CustomProcessingBlock((f, src) =>
                {
                    using (var releaser = new FramesReleaser())
                    {
                        foreach (ProcessingBlock p in blocks)
                            f = p.Process(f).DisposeWith(releaser);
                        var decimationFilter = new DecimationFilter();
                        var DisparityTransform = new DisparityTransform();
                        var spatialFilter = new SpatialFilter();
                        var temporalFilter = new TemporalFilter();
                        var holeFillingFilter = new HoleFillingFilter();
                        var thresholdFilter = new ThresholdFilter();
                        
                        f = f.ApplyFilter(decimationFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(DisparityTransform).DisposeWith(releaser);
                        f = f.ApplyFilter(spatialFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(temporalFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(holeFillingFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(thresholdFilter).DisposeWith(releaser);

                        f = f.ApplyFilter(align).DisposeWith(releaser);
                        f = f.ApplyFilter(colorizer).DisposeWith(releaser);

                        var frames = f.As<FrameSet>().DisposeWith(releaser);

                        var colorFrame = frames[Stream.Color, Format.Rgb8].DisposeWith(releaser);
                        var colorizedDepth = frames[Stream.Depth, Format.Rgb8].DisposeWith(releaser);

                        var res = src.AllocateCompositeFrame(colorizedDepth, colorFrame).DisposeWith(releaser);
                        src.FrameReady(res);
                    }
                });`
                影片

https://github.com/IntelRealSense/librealsense/assets/134619419/159ed557-d1bf-4b8f-a098-2a0af866b8c2

MartyG-RealSense, 2023-08-22:
You do not need to use every filter type that is listed in the docuentation, only the types that you need.  Because post processing filters are processed on the CPU of the computer and not on the camera hardware, using a lot of filters can degrade the performance of your program.

The listed order of filters is intended to show which filters should be listed before and after others.  So if you only wanted to apply Decimation, Spatial and Temporal filters then they should be applied in this order:

Decimation
Spatial
Temporal

Most RealSense users will not use the Depth2Disparity and Disparity2Depth filters.

*********

您不需要使用文檔中列出的每種過濾器類型，只需使用您需要的類型。 由於後處理濾鏡是在計算機的 CPU 上處理的，而不是在相機硬件上處理的，因此使用大量濾鏡會降低程序的性能。

過濾器的列出順序旨在顯示哪些過濾器應列在其他過濾器之前和之後。 因此，如果您只想應用抽取、空間和時間濾波器，那麼應​​按以下順序應用它們：

抽取
空間
顳

大多數 RealSense 用戶不會在其腳本中使用 Depth2Disparity 和 Disparity2Depth 過濾器。

ned6602, 2023-08-22:
我發現我的深度值只在0~255之間跳動,請問是哪裡出問題了
```
block.Start(f =>
                {
                    using (var frames = f.As<FrameSet>())
                    {
                        var colorFrame = frames.ColorFrame.DisposeWith(frames);
                        var colorizedDepth = frames.First<VideoFrame>(Stream.Depth, Format.Rgb8).DisposeWith(frames);

                        var imageColor = new Image<Rgb, byte>(colorFrame.Width, colorFrame.Height);
                        var imageDepth = new Image<Rgb, byte>(colorizedDepth.Width, colorizedDepth.Height);

                        var colorFrameBytes = new byte[colorFrame.Stride * colorFrame.Height];
                        var colorizedDepthBytes = new byte[colorizedDepth.Stride * colorizedDepth.Height];

                        System.Runtime.InteropServices.Marshal.Copy(colorFrame.Data, colorFrameBytes, 0, colorFrameBytes.Length);
                        System.Runtime.InteropServices.Marshal.Copy(colorizedDepth.Data, colorizedDepthBytes, 0, colorizedDepthBytes.Length);

                        imageColor.Bytes = colorFrameBytes;
                        imageDepth.Bytes = colorizedDepthBytes;

                        // 將彩色影像轉換為灰度影像進行處理
                        var grayImage = imageColor.Convert<Gray, byte>();

                        //進行高斯模糊以減少噪音（可選）
                        CvInvoke.GaussianBlur(grayImage, grayImage, new Size(5, 5), 0);

                        // 對灰度影像進行二值化處理
                        var binaryImage = grayImage.ThresholdBinary(new Gray(240), new Gray(255));

                        // 在二值影像中尋找輪廓
                        VectorOfVectorOfPoint contours = new VectorOfVectorOfPoint();
                        CvInvoke.FindContours(binaryImage, contours, null, RetrType.List, ChainApproxMethod.ChainApproxSimple);

                        // 找到最大的輪廓
                        int largestContourIndex = 0;
                        double largestContourArea = 0;
                        for (int i = 0; i < contours.Size; i++)
                        {
                            double contourArea = CvInvoke.ContourArea(contours[i]);
                            if (contourArea > largestContourArea)
                            {
                                largestContourArea = contourArea;
                                largestContourIndex = i;
                            }
                        }

                        // 在影像上繪製邊界矩形和中心像素座標
                        var largestContour = contours[largestContourIndex];
                        var boundingRect = CvInvoke.BoundingRectangle(largestContour);
                        Point centerPixel = new Point(boundingRect.X + boundingRect.Width / 2, boundingRect.Y + boundingRect.Height / 2);

                        // 在影像上顯示中心像素座標
                        CvInvoke.PutText(imageColor, $"Center: ({centerPixel.X},{centerPixel.Y})", new Point(centerPixel.X -100, centerPixel.Y - 10), FontFace.HersheyComplex, 2.0, new MCvScalar(0, 0, 255), 2);
                        
                        // 在影像上繪製邊界矩形
                        CvInvoke.Rectangle(imageColor, boundingRect, new MCvScalar(0, 0, 255), 10);

                        // 複製深度數據
                        byte[] depthData = new byte[colorizedDepthBytes.Length];
                        colorizedDepth.CopyTo(depthData);

                        // 取得中心點對應的深度值
                        int depthIndex = centerPixel.Y * colorizedDepth.Width + centerPixel.X;
                        ushort depthValue = depthData[depthIndex];
                        SharedVariables.DepthValue = depthValue; // 在 SharedVariables 中設定深度值
                        CvInvoke.PutText(imageColor, $"Depth: {depthValue:F3} mm", new Point(centerPixel.X + 10, centerPixel.Y + 50), FontFace.HersheyComplex, 2, new MCvScalar(0, 0, 255), 2);

                        // 在 SharedVariables 中設定中心像素座標
                        SharedVariables.CenterPixel = centerPixel;
                        

                        // 更新 imageBoxColor
                        imageBoxColor.Image = imageColor;                        

                        // 在 Emgu.CV.UI.ImageBox 控制項中顯示深度影像
                        imageBoxDepth.Image = new Image<Rgb, byte>(imageDepth.Bitmap);
                        
                    }
                });
```

MartyG-RealSense, 2023-08-22:
RealSense raw pixel depth values are between 0 and 65535.  When you convert a RealSense image to an OpenCV one, the range changes to between 0 and 255.

*******

RealSense 原始像素深度值介於 0 到 65535 之間。將 RealSense 圖像轉換為 OpenCV 圖像時，範圍會更改為 0 到 255 之間。

MartyG-RealSense, 2023-08-29:
Do you require further assistance with this case, please?  Thanks!

請問您對此案需要進一步的幫助嗎？ 謝謝！

ned6602, 2023-08-29:
我目前有成功獲取深度圖與正確的深度值了 ,
現在我想要做Eye in hand 的手眼校正. 請問要怎麼透過Realsense 獲得標定版的RX RY RZ?

MartyG-RealSense, 2023-08-30:
There are hand-eye calibration tools for robot arms that work with RealSense 400 Series cameras.  Here are a couple of suggestions.

**autoCalibration**
https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-475621533 

**'Robotic Hand-eye Calibration Workspace' eye-in-hand for ROS**
https://github.com/lixiny/Handeye-Calibration-ROS 

The above system makes use of **visp**, which has been separately recommended by some for hand-eye calibration with RealSense.  http://wiki.ros.org/visp_hand2eye_calibration 

Another example of a hand-eye calibration tool that can be used with RealSense is **MoveIt**.

https://github.com/ros-planning/moveit_calibration

***************

有適用於與 RealSense 400 系列相機配合使用的機器人手臂的手眼校準工具。 這裡有一些建議。

**自動校準**
https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-475621533

**ROS 的“機器人手眼校準工作空間”手眼協作**
https://github.com/lixiny/Handeye-Calibration-ROS

上述系統使用了 **visp**，一些人單獨推薦它用於 RealSense 手眼校準。 http://wiki.ros.org/visp_hand2eye_calibration

可與 RealSense 一起使用的手眼校準工具的另一個示例是 **MoveIt**。

https://github.com/ros-planning/moveit_calibration

ned6602, 2023-09-06:
目前沒有,感謝您的幫助!!

MartyG-RealSense, 2023-09-06:
You are very welcome.  Thanks very much for the update!  As you do not require further assistance at this time, I will close this case.  Thanks again!

非常歡迎您。 非常感謝您的更新！ 由於您目前不需要進一步的幫助，我將關閉此案例。 再次感謝！

closed 2023-09-06 by MartyG-RealSense

m1234567891111, 2025-02-19:
> RealSense raw pixel depth values are between 0 and 65535. When you convert a RealSense image to an OpenCV one, the range changes to between 0 and 255.
> 
> RealSense 原始像素深度值介於 0 到 65535 之間。將 RealSense 圖像轉換為 OpenCV 圖像時，範圍會更改為 0 到 255 之間。

I would like to ask how to fix it in the end

m1234567891111, 2025-02-19:
> 成功獲取深度圖與正確的深度值了

How to Successfully obtained the depth map and correct depth value
i need your help thanks !

ned6602, 2025-02-19:
> > 成功獲取深度圖與正確的深度值了
> 
> How to Successfully obtained the depth map and correct depth value i need your help thanks !

var colorizedDepth = frames.First<VideoFrame>(Stream.Depth, Format.Rgb8).DisposeWith(frames);
change "Format.Rgb8" to "Format.Z16"
Thanks!

ned6602, 2025-02-19:
> > 成功獲取深度圖與正確的深度值了
> 
> How to Successfully obtained the depth map and correct depth value i need your help thanks !
add depthFrame and use Format.Z16
Thanks!
`using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
using System.Threading;
using System.Threading.Tasks;
using System.Windows.Forms;
using Intel.RealSense;
using System.Windows.Media.Imaging;
using AForge.Imaging;
using AForge.Imaging.Filters;
using AForge.Math.Geometry;
using System.Drawing.Imaging;

namespace WindowsFormsApp3
{
    public partial class Form1 : Form
    {
        Intel.RealSense.Context ctx = new Intel.RealSense.Context();
        private Pipeline pipeline = new Pipeline();
        private Colorizer colorizer = new Colorizer();
        private Align align = new Align(Stream.Color);
        private CustomProcessingBlock block;
        private CancellationTokenSource tokenSource = new CancellationTokenSource();

        static Action<VideoFrame> UpdateImage(PictureBox pictureBox)
        {
            return new Action<VideoFrame>(frame =>
            {
                var bitmap = new System.Drawing.Bitmap(frame.Width, frame.Height, frame.Stride, System.Drawing.Imaging.PixelFormat.Format24bppRgb, frame.Data);
                pictureBox.Image = bitmap;
                pictureBox.Refresh();
            });
        }

        private List<System.Windows.Point> FindCircleCenters(Bitmap bitmap)
        {
            var image = new Bitmap(bitmap);

            // Perform circle detection on the image
            var gray = Grayscale.CommonAlgorithms.BT709.Apply(image);
            var filter = new Threshold(150);
            filter.ApplyInPlace(gray);

            var blobCounter = new BlobCounter();
            blobCounter.FilterBlobs = true;
            blobCounter.MinHeight = 5;
            blobCounter.MinWidth = 5;
            blobCounter.ProcessImage(gray);

            var blobs = blobCounter.GetObjectsInformation();
            List<System.Windows.Point> centers = new List<System.Windows.Point>();
            foreach (var blob in blobs)
            {
                var center = blob.CenterOfGravity;
                var wpfCenter = new System.Windows.Point(center.X, center.Y);
                centers.Add(wpfCenter);
            }

            return centers;
        }

[... 127 lines cut ...]

                        var bitmapColor = new Bitmap(colorFrame.Width, colorFrame.Height, colorFrame.Stride, PixelFormat.Format24bppRgb, colorFrame.Data);
                        var bitmapDepth = new Bitmap(colorizedDepth.Width, colorizedDepth.Height, colorizedDepth.Stride, PixelFormat.Format24bppRgb, colorizedDepth.Data);

                        var circleCenters = FindCircleCenters(bitmapColor);
                        var depthValues = GetDepthValues(depthFrame, circleCenters);
                        DrawNineCircleCenters(bitmapColor, circleCenters, depthValues);

                        pictureBoxDepth.Invoke(updateDepth, colorizedDepth);
                        pictureBoxColor.Invoke(updateColor, colorFrame);
                    }
                });
            }
            catch (Exception ex)
            {
                MessageBox.Show("An error occurred: " + ex.Message);
            }
        }

        private void Form1_FormClosing(object sender, FormClosingEventArgs e)
        {
            tokenSource.Cancel();
        }
    }
}`

ned6602, 2025-02-19:
> > 成功獲取深度圖與正確的深度值了
> 
> How to Successfully obtained the depth map and correct depth value i need your help thanks !

Add new depthFrame and use Format.Z16
`var depthFrame = frames[Stream.Depth, Format.Z16].DisposeWith(releaser)`
Thanks!
[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/tutorial](url)

ned6602, 2025-02-19:
> > 成功獲取深度圖與正確的深度值了
> 
> How to Successfully obtained the depth map and correct depth value i need your help thanks !
Add depthFrame and use Format.Z16
` var depthFrame = frames[Stream.Depth, Format.Z16].DisposeWith(releaser)`
you can refer
 [[librealsense](https://github.com/IntelRealSense/librealsense/tree/master)wrappers/csharp](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp)

MartyG-RealSense, 2025-02-19:
Hi @m1234567891111  There are very few C# references available for converting RealSense rs2::frame to OpenCV's cv::mat and obtaining the depth value from the mat, unfortunately.  A C# script at https://github.com/IntelRealSense/librealsense/issues/8226 may be a helpful reference for doing so.

MartyG-RealSense, 2025-02-19:
@ned6602  Thanks so much for your own advice to @m1234567891111

m1234567891111, 2025-02-24:
I am currently facing a problem. After making corrections, the depth values can now be displayed.
However, I need the color image on the right side, but it has turned into a grayscale image.
How should I fix this?

[image]
[image]

m1234567891111, 2025-02-24:
I measured data with the D435 camera, and the results are more accurate than those from the L515.
What could be the problem?

700mm that picture is D435

[image]
[image]

MartyG-RealSense, 2025-02-24:
D435 and L515 use different camera technologies (D435 is Stereo Depth and L515 is Lidar Depth) so the results can differ depending on the kind of surface that is being observed by the camera.  For example, L515 has more difficulty with depth-sensing transparent plastic drinks bottles than D435.

A method that often improves depth results on L515 is to set its Visual Preset camera configuration to **Short Range** instead of the default Max Range.  The Short Range preset lowers the values of the L515's 'Laser Power' and 'Receiver Gain' settings.  These settings could also be configured individually with program code instead of using a Visual Preset.