realsenseai/librealsense #12122: Left IC2 Config error

hermattt, 2023-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 5.15.0.2| 
| Operating System & Version | Win 11 | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.54.1                           | 
| Language                            |  -                          | 
| Segment			|  Robot | 

### Issue Description
Hi, I previously used Intel RealSense Viewer, but recently it stopped working. 
Error in console: 
>[Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:511 - Left IC2 Config error

When turning on streaming there is "No Frames Received" info.
Updating the firmware didn't help, When trying "Hardware Reset", console shows:
>[Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:661 - acquire_power failed: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "The system cannot find the path specified." 
[Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\error-handling.cpp:96 - Error during polling error handler: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "The system cannot find the path specified."

MartyG-RealSense, 2023-08-22:
Hi @hermattt  Can you check the camera privacy settings in Windows please to ensure that pernission for applications to access the camera is set to On.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

hermattt, 2023-08-22:
[image]
It was set to on.

MartyG-RealSense, 2023-08-22:
If you are using a D410 depth module, please next check that the interposer (the floppy cable or rigid board that joins the Vision Processor D4 and Depth Module together) is fully seated in the 50-pin connectors and is not only partially inserted.

hermattt, 2023-08-22:
It is fully seated. (Reconnected, no difference)

hermattt, 2023-08-22:
Commercial flexible cable made by Kinova (their Gen3 robot). 
[image]

When it was working, I installed old (in that time it was new), but now tried it and it didn't work. I updated to newer release using Intel.RealSense.Viewer.exe version. 

Installing new version Intel.RealSense.SDK-WIN10 did not help.

rs-hello-realsense.exe opens console for 20 seconds and exits with no output.

MartyG-RealSense, 2023-08-22:
What is the product in the picture above, please?  Is it a RealSense D415 camera that has been opened and its camera board disconnected?

MartyG-RealSense, 2023-08-22:
Thanks very much for the confirmation that it is an opened-up Kinova Gen 3 with a D410 depth module.

The information listing in the Viewer looks fine.  

Kinova Gen 3 can work with the RealSense SDK, as demonstrated by the case at https://github.com/IntelRealSense/librealsense/issues/11993 and your own experiences with the Viewer.

Intel are unable to provide technical support though for products not manufactured by Intel that incorporate RealSense technology, unfortunately.  The proper place to seek support for such products will be the manufacturer, which in this instance is Kinova.

https://www.kinovarobotics.com/reach-us/customer-support

It looks as though the Intel firmware driver enables the Vision Processor and D410 camera boards to be recognized by Intel software such as the Viewer.  It likely would not recognize the particular Omnivision RGB sensor that Kinova are using as it is a different model from the Omnivision sensors used for RGB by Intel-made RealSense cameras, which is why the firmware identifies it as D410 (depth and infrared streams only, no RGB stream).

In summary, Intel cannot provide replacement components if there is a hardware fault because it is a Kinova product.  Kinova might not replace the parts because if the Kinova was still within its sales warranty period then opening it up would have likely invalidated the warranty.

I would recommend reconnecting the D410 board to the interposer cable and then trying to run a **calibration** of the sensors to see if it clears the problem that you are experiencing (though the calibration will probably fail if streaming cannot be started).   If that does not work then purchasing replacement Vision Processor D4 V1 / RealSense D410 boards might be the only solution.  I think that the problem is more likely to be with the D410 board than the Vision Processor D4, as there is not an apparent problem with accessing the camera details stored on the D4.

MartyG-RealSense, 2023-08-29:
Hi @hermattt  Do you have an update about this case that you can provide, please?  Thanks!

hermattt, 2023-08-31:
Looking under microscope there are 2 elements that seem to be broken on D4 card.
[image]
[image]
I`m waiting for D4 board, if it won't help, I will buy camera module. 
Is there any board view of that board so I could check what are those damaged components?

MartyG-RealSense, 2023-08-31:
There is not a fully annoted diagram available of the components on the D4 V1 board.  The main components of the V1 though will be the D4 chip, an EEPROM flash memory component for storing the camera firmware driver, the USB components and an Image Signal Processor (ISP) chip for handling color data processing.

[image]

hermattt, 2023-08-31:
So, I can't get to know, what those red marked components are? 
[image]

MartyG-RealSense, 2023-08-31:
There is not an available diagram that identifies those parts, unfortunately.

closed 2023-08-31 by hermattt