realsenseai/librealsense #12123: Issue with Hard Crashes on Realtime Linux Kernel + ROS Melodic

jwatson-CO-edu, 2023-08-21:
| Required Info                         |                                                            |
|----------------------------------------|------------------------------------------------- |
| Camera Model                       | D405                                                   | 
| Firmware Version                  | 5.15.0.2                                               | 
| Operating System & Version |   Ubuntu 18.04                                    | 
| Kernel Version                       |  4.19.50-rt22, Realtime Kernel Patch | 
| Platform                                 | PC, Xeon E3-1200                             |
| SDK Version                          | v2.54.1, Commit # 8ffb17b0               | 
| Language                              |  C++, ROS Melodic                            | 
| Segment			        |  Robot                                                 | 

I am having an issue with a node on ROS Melodic based on the librealsense SDK on Ubuntu 18.04 with 4.19.50-rt22 realtime kernel patch.  The node itself is a light wrapper that packages RealSense data as ROS messages.  
After anywhere between 5 and 50 minutes of starting the node, my computer will crash; the UI will freeze and a few seconds later the OS will restart entirely. (I tracked it down to this node by process of elimination.)
Neither dmesg nor journalctl reveal anything particularly useful about the crash. Data coming from the D405 appears correct and arrives in a timely manner.  

I built librealsense from source with RSUSB based on the idea that this might support older kernels, **but the problem persists**.

I used the following steps to build librealsense:  
1. `sudo apt install libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev cmake`
1. `sudo apt install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at`
1. `cd /tmp/`
1. `git clone https://github.com/IntelRealSense/librealsense.git`
1. `cd librealsense/`
1. Unplug any attached RealSense cameras
1. `./scripts/setup_udev_rules.sh`
1. `./scripts/patch-realsense-ubuntu-lts.sh`
1. Add the following near the top of CMakeLists.txt: `set( FORCE_RSUSB_BACKEND ON )`
1. `mkdir build && cd $_`
1. `cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true`
1. `sudo make uninstall && make clean && make && sudo make install`  

**Can you think of anything in particular about librealsense that may interfere/interact with realtime scheduling?**

jwatson-CO-edu, 2023-08-22:
1. I use the ROS1 wrapper that you mentioned, but thank you for the suggestion, @MartyG-RealSense.
2. I made the changes to the build process that you suggested, and librealsense built successfully, but the problem persists.  

**Does librealsense perform any logging that would remain after a restart that I can use for troubleshooting?**  

This bug is currently hampering my entire ROS stack.

MartyG-RealSense, 2023-08-24:
Linux writes all OS logs to the  **/var/log** folder.   You can review the entire kernel log file with the command below.

`less /var/log/kern.log`

jwatson-CO-edu, 2023-08-27:
This was an unknown hardware issue that resolved itself when I replaced the computer I was using with an identical machine.
The problem is not present on the new computer and I was able to run programs using librealsense both inside and outside of ROS for hours at a time with no issue.  
Troubleshooting did not reveal the problem on the original computer, but evidence does not point to a problem with this library.  
Issue closed.

closed 2023-08-27 by jwatson-CO-edu