realsenseai/librealsense #12129: Recording depth and colour frames

Susheelabhat, 2023-08-23:
| Camera Model D435i/D455
| Firmware Version
| Operating System & Version Ubuntu 17)
| Kernel Version (Linux Only)
| Platform Raspberry Pi
| SDK Version 2.0
| Language python
| Segment others

Hi, I want to record bag files and I was doing it at 6fps with a lower resolution using Realsense-viewer app.
I want to do it without the application now. Is there a possibility of achieving this through command line? 
The reason is I'm using a raspberry PI to which I'm not connecting a monitor/screen , so I have to remotely start the recording and change the parameters.

Thanks in advance!

MartyG-RealSense, 2023-08-24:
Hi @Susheelabhat  At the empty line 33 of the **rs-record** program, you could insert additional 'cfg' configurations line to set the minimum FPS speed for the depth and color streams.

https://github.com/IntelRealSense/librealsense/blob/master/tools/recorder/rs-record.cpp#L33

cfg.enable_stream(RS2_STREAM_COLOR, 848,480, RS2_FORMAT_Z16, 6);
cfg.enable_stream(RS2_STREAM_COLOR,1280,720, RS2_FORMAT_RGB8, 6);

For the D455 camera model, the FPS should be set to 5 instead of 6 as D455 does not support 6 FPS.

If you wish to use a different resolution then also edit the resolution in the cfg lines.  For example, to set both streams to 640x480:

```
cfg.enable_stream(RS2_STREAM_COLOR, 640,480, RS2_FORMAT_Z16, 6);
cfg.enable_stream(RS2_STREAM_COLOR, 640,480, RS2_FORMAT_RGB8, 6);
```

Susheelabhat, 2023-08-25:
Thanks for the info @MartyG-RealSense. Wanted to check one more thing, 
In the ubuntu system, what is the default location of these librealsense tools/files where I can make changes?

MartyG-RealSense, 2023-08-25:
Rather than looking in the 'build' folder, the appropriate location in the source code folder is likely to be **librealsense > tools > recorder** so that you can edit the contents of the rs-record.cpp file and **then** build librealsense from that modified source code.

[image]

Susheelabhat, 2023-08-30:
I'll try this @MartyG-RealSense . Another follow up question, with the recorded .bag file, is it possible to extract depth frames outside ubuntu(realsense sdk) (where pyrealsense exists)?
I was using this code to extract frames. But how can I do it without pyrealsense support?
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/read_example_bag.py

MartyG-RealSense, 2023-08-30:
You can run a tool provided by the RealSense SDK called **rs-convert** to extract depth frames from a bag file recorded in the SDK.  Though it is made from C++ code, it is available pre-built so that you can launch the tool without having to compile the code.

If you included the examples and tools with the flag **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** when building librealsense with CMake then rs-convert should be in the **usr/local/bin** folder of Ubuntu.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

Susheelabhat, 2023-09-01:
I tried building librealsense  again with the code added in rs-recorder.cpp. But when I now run the command  **rs-record -f ./test1.bag -t 60**  , I'm getting segmentation fault error. 

pi@pi2:~/Documents$ rs-record -f /home/pi/Documents/test.bag -t 60
Segmentation fault (core dumped)

MartyG-RealSense, 2023-09-02:
Was the added code the cfg lines to set 6 FPS?

closed 2023-09-15 by MartyG-RealSense