realsenseai/librealsense #12130: Customize MSVC_RUNTIME_LIBRARY  in realsense-filed.lib

frbitten, 2023-08-23:
I'm compiling the SDK with cmake and CL (Visual Studio build tools). I have an MSVC_RUNTIME_LIBRARY incompatibility problem where realsense-filed.lib is being compiled with MTd_StaticDebug and I would need to change to MDd_DynamicDebug.

I didn't find any option to configure via option in cmake. How should I go about changing the build option.
I use ExternalProject_Add to add the SDK dependency to my project.

```[build] realsense-filed.lib(bag.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(duration.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(message_instance.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(header.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(query.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(view.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(stream.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(uncompressed_stream.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj [E:\engine\PA_Engine\EngineEcosystem\Engine\build\niopEngine.vcxproj]
[build] realsense-filed.lib(lz4_stream.obj) : error LNK2038: não correspondência detectada para 'RuntimeLibrary': valor 'MTd_StaticDebug' não corresponde ao valor 'MDd_DynamicDebug' no CommunicationsManager.obj ```

Best Regards
Fernando Bittencourt

frbitten, 2023-08-24:
I found a way to configure using the BUILD_WITH_STATIC_CRT variable. I couldn't find this variable anywhere in the documentation. Luckily I bumped into it when analyzing the code.

closed 2023-08-24 by frbitten

MartyG-RealSense, 2023-08-24:
It's great to hear that you achieved a solution, @frbitten - thanks very much for sharing it!