realsenseai/librealsense #12131: Unsupported JetPack revision 35.3.1 aborting script

jidnyesha-patil, 2023-08-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Linux L4T (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                         | 
| Platform                                 | NVIDIA Jetson |
| Language                            |  python                    | 
| Segment			|  Robot                          | 

### Issue Description
Trying to install librealsense from source to solve pyrealsense2 issues. I accidentally deleted the rules file before so installing librealsense from scratch again. When trying to apply patched kernel module for Jetson L4T, I get the error "Unsupported JetPack revision 35.3.1 aborting script"
[image]

Based on Issue https://github.com/IntelRealSense/librealsense/issues/11391#issue-1569854831, I need to match the supported JetPack version. Looking into the patch files, supported versions shown below.
[image]
Highest supported patch is 35.1 while I have 35.3.1.
Need suggestions on how to downgrade since there have multiple issues which say downgrading is not possible. Any help would be appreciated.

MartyG-RealSense, 2023-08-24:
Hi @jidnyesha-patil  You should be able to downgrade by flashing the JetPack using the **Nvidia SDK Manager** tool, as described at the link below.

https://forums.developer.nvidia.com/t/downgrade-jetpack/169373/3

Alternatively, you could build librealsense from source in **RSUSB backend** mode by including the flag **-DFORCE_RSUSB_BACKEND=TRUE** in your CMake build instruction.  An RSUSB build of librealsense bypasses the kernel.

jidnyesha-patil, 2023-08-24:
Got it! I am trying the RSUSB backend mode first since I must get SDK manager to allow flashing JetPack.I am currently using the pre-flash SD card that came with the kit. Will keep this thread updated. 
Thank you @MartyG-RealSense

dmipx, 2023-08-28:
@jidnyesha-patil Try to append "35.3.1" to "35.1", like: `"35.1 | 35.3.1")`

jidnyesha-patil, 2023-08-29:
The RSUSB mode installation worked!

closed 2023-08-29 by jidnyesha-patil

MartyG-RealSense, 2023-08-30:
That's great to hear that RSUSB was successful for you, @jidnyesha-patil - thanks very much for the update!