realsenseai/librealsense #12133: Unable to compile my capture.cpp via cmake(Clion)

No-9527149, 2023-08-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Win10  | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC  |
| SDK Version                            | 2.54.1                       | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description

Dear developers, when I use CLion and CMakeLists.txt to compile my own capture.cpp, I encountered a problem that it cannot be compiled.

First encountered compile error as shown below:

```
FAILED: test.exe 
cmd.exe /C "cd . && 
D:\mingw64\bin\c++.exe -g  CMakeFiles/test.dir/capture.cpp.obj -o test.exe -Wl,--out-implib,libtest.dll.a -Wl,--major-image-version,0,--minor-image-version,0 
-LC:/PROGRA~2/INTELR~1.0/bin/x64 -lrealsense2  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
CMakeFiles/test.dir/capture.cpp.obj: In function `set_viewport(rect const&)':
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:135: undefined reference to `__imp_glViewport'
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:136: undefined reference to `__imp_glLoadIdentity'
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:137: undefined reference to `__imp_glMatrixMode'
...
CMakeFiles/test.dir/capture.cpp.obj: In function `window::window(int, int, char const*)':
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:524: undefined reference to `glfwInit'
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:525: undefined reference to `glfwCreateWindow'
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:528: undefined reference to `glfwMakeContextCurrent'
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:530: undefined reference to `glfwSetWindowUserPointer'
C:/Program Files (x86)/Intel RealSense SDK 2.0/samples/example.hpp:531: undefined reference to `glfwSetMouseButtonCallback'
```

I checked the `example.hpp` file, a series of functions such as `glViewport` can be found in `gl.h` of `mingw`, functions such as `glfwInit` can also be found in `glfw3.h` of `mingw`. 

How can I fix this error and implement `capture.cpp` using CMake on CLion?

Below is my `CMakeList.txt`
```
cmake_minimum_required(VERSION 3.26)
project(test)

set(CMAKE_CXX_STANDARD 17)

include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/include")
include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/samples")
include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/third-party")

link_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/bin/x64")

add_executable(test capture.cpp)

target_link_libraries(test realsense2.dll)
```

No-9527149, 2023-08-24:
After installing `OpenGL`, `FreeGLUT`,`GLEW`, `GLFW`, I got `libopengl32.a`, `libglu32.a`, `libglfw3.a` in mingw's lib, and then added these three .a files to CMakeLists.txt, I successfully ran own `capture.cpp`.

Interestingly, Intel not only wants to promote its own DX, but also uses OpenGL in this project, but while using OpenGL, it does not fully compile the relevant library files. This thing tormented me for two days, thanks a lot for the Intel RealSense SDK.

Below is my new `CMakeList.txt`:

```
cmake_minimum_required(VERSION 3.26)
project(test)

set(CMAKE_CXX_STANDARD 17)

include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/include")
include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/samples")
include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/third-party")

link_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/bin/x64")

add_executable(test capture.cpp)

target_link_libraries(test realsense2.dll libglfw3.a libopengl32.a libglu32.a)
```

closed 2023-08-24 by No-9527149

MartyG-RealSense, 2023-08-24:
Thanks so much @No-9527149 for sharing the details of your solution!