realsenseai/librealsense #12136: Connecting Intel realsense to Xilinx KV260

Nimaro76, 2023-08-25:
Hello ,

I have an issue that I would appreciate it if you could guide me. I need to connect D435F to Xilinx KV260 to use the Python or c++ algorithms. However, despite the fact that I followed the SDK installation, rs-enumerate-devices shows no device is plugged in . I could not install the kernel patch step though but I did have permission for recognition of the camera . The board has Ubuntu 20 focal image 
And it has arm core processor. I could not install pyrealsense through pip either. 
thank you in advance for your help.
Kind regards,

dmipx, 2023-08-28:
Hi @Nimaro76 
You can verify if device enumerated by running `lsusb`
In case you will have another camera or media controller on-board, you will have such issue as camera detection by SDK.
We are working on solution for multiple media controllers, track this pr: #12060 
You can verify and share your platform setup:
`v4l2-ctl --list-devices`

MartyG-RealSense, 2023-09-04:
Hi @Nimaro76  Do you require further assistance with this case, please?  Thanks!

Nimaro76, 2023-09-04:
Thank you for your support . Not for the moment.

MartyG-RealSense, 2023-09-04:
@Nimaro76  You are very welcome.  Thanks very much for the update!

@dmipx  Thanks for your help in this case!

Nimaro76, 2023-09-06:
Hello, 
I ran into a problem that I would appreciate your help . As advised, I installed the SDK with RSUSB backend . However, data transfer appears to be very slow. for instance this part of code takes 1.2 sec: 

      rs_frames = pipeline.wait_for_frames()   
        aligned_frames = align.process(rs_frames)
        rs_depth_frame = aligned_frames.get_depth_frame()
        rs_color_frame = aligned_frames.get_color_frame()
        # Convert images to numpy arrays
        depth_image = np.asanyarray(rs_depth_frame.get_data())
        color_image = np.asanyarray(rs_color_frame.get_data())

        # use YOLO V8 to segment
        timestamp1=int(time.time()*1000)

        results = model.predict(color_image, show=False)

        timestamp2=int(time.time()*1000)
        timestamp=timestamp2-timestamp1
        print(timestamp)

could you please tell me whether RSUSB has slowed down the data transfer ? I also see the delay in data in realsense-viewer installed on ubuntu image of KV260 as well . 
thank you for your help .

MartyG-RealSense, 2023-09-07:
An RSUSB build of the SDK will typically react slower than a kernel-patched build when changing options from one value to another (such as changing exposure).  That does not seem to be the case in this situation though as you are not changing option values.

Alignment is performed on the computer's processor rather than the camera hardware and can be a processing-expensive operation.  If you are also running YOLO at the same time then that is also likely placing a burden on the CPU.

Nimaro76, 2023-09-07:
thank you for your response. do you have any suggestions regarding possible fixes ? I was really surprised to see that even realsense-viewer is slow in KV260.

MartyG-RealSense, 2023-09-07:
If you are using RGB and are building the SDK from source code then you may be able to reduce latency by including the build flag `-DBUILD_WITH_OPENMP=TRUE` in the CMake build instruction.  This enables the SDK to take advantage of multiple CPU cores when processing RGB instead of the single CPU core that the SDK uses by default.

closed 2023-09-19 by MartyG-RealSense