realsenseai/librealsense #12137: glbic_2.29

naorwaiss, 2023-08-25:
hi this is my problem 

/home/fhsbs/.virtualenvs/pythonProject2/bin/python /tmp/pycharm_project_762/depth_camera.py 
Traceback (most recent call last):
  File "/tmp/pycharm_project_762/depth_camera.py", line 4, in <module>
    import pyrealsense2 as rs
  File "/home/fhsbs/.local/lib/python3.9/site-packages/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ImportError: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/fhsbs/.local/lib/python3.9/site-packages/pyrealsense2/pyrealsense2.cpython-39-aarch64-linux-gnu.so)

Process finished with exit code 1

i try to run simple python code 

python version 3.9.12
pyrealsense2  2.54.1.5217

i do the download from the clone (ask chat gpt for the download)
did someone can help me 

(got ld515 cam)

MartyG-RealSense, 2023-08-26:
Hi @naorwaiss  This error has not previously been reported in regard to the pyrealsense2 wrapper.

As aarch64 is referenced, it looks as though your computer / computing device is based on the Arm hardware architecture.  So that means that pyrealsense2 has to be built from source code, as the **pip install pyrealsense2** package install method will not work on Arm.

Which computer / computing device are you using, please?

pyrealsense2 can be installed from source with CMake at the same time as building the librealsense SDK or afterwards.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 has a guide for installing librealsense and pyrealsense2 together on Nvidia Jetson (an Arm-based board) that you may be able to adapt for your own compuer if you are not using Jetson.

In the `-DPYTHON_EXECUTABLE` instruction, change python3.6 to python3.9.

Alternatively, the method for installing pyrealsense2 after librealsense has already been installed is described in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md#building-from-source

naorwaiss, 2023-08-26:
hi, i am using jetson nano

MartyG-RealSense, 2023-08-26:
The https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 guide should be ideal for you then as it was written particularly for Jetson Nano (though it works with most Jetson board models).

naorwaiss, 2023-08-26:
hi... its didnt work..
i try use his version but didnt work 
when i use pip instal pyrealsese the glib is back 
when i didnt do the pip install and do the install manualy like what u send the python didnt recugnize the pyrealsense

thanks alot

MartyG-RealSense, 2023-08-27:
I noe that you are using pycharm.  There is a past case at https://github.com/IntelRealSense/librealsense/issues/4089 where a Nano user with pycharm experienced the error **ModuleNotFoundError: No module named 'pyrealsense2'** when, like you, they simply ran the line the pyrealsense2 import instruction.  Another RealSense user in that case suggested at https://github.com/IntelRealSense/librealsense/issues/4089#issuecomment-496988560 to edit the **bashrc file**.

naorwaiss, 2023-08-27:
o try it also.
let me understand.. when i do the download of the realsense in the command i need to do also pip install librealsense2?
or when i do all the donlwad: cmake, SDK, etc at the end i need make path drom the python to the librealsense witouth the pip install?

MartyG-RealSense, 2023-08-27:
**pip install pyrealsense2** cannot be used on Jetson Nano as this installation method is not compatible with Arm devices such as Nvidia Jetson.  It should not be used at all on Nano.

naorwaiss, 2023-08-28:
ok have some step foword today 

/home/fhsbs/.virtualenvs/pythonProject2/bin/python /tmp/pycharm_project_260/depth_camera.py 
Traceback (most recent call last):
  File "/tmp/pycharm_project_260/depth_camera.py", line 13, in <module>
    pipeline = rs.pipeline()
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

Process finished with exit code 1

now this problem is rising

MartyG-RealSense, 2023-08-29:
Does the error still occur if you input the lines below into the terminal?

```
export PATH=$PATH:~/.local/bin
export PYTHONPATH = $PYTHONPATH:/usr/local/lib/python3.9/pyrealsense2
```

MartyG-RealSense, 2023-08-29:
Instead of using **import pyrealsense2 as rs**, is the problem corrected if you instead use this: 

**import pyrealsense2.pyrealsense2 as rs**

naorwaiss, 2023-08-30:
thanks bro its help i got camera and depth with the pyrealsense2

MartyG-RealSense, 2023-08-30:
You are very welcome.  That's great to hear that you were successful!

MartyG-RealSense, 2023-09-11:
Case closed due to no further comment received.

closed 2023-09-11 by MartyG-RealSense