realsenseai/librealsense #12139: Intel L515 low (~4hz) RGB framerate

brtmax, 2023-08-27:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-78-generic                           | 
| Platform                                 | Intel NUC |
| SDK Version                            | 2.50.0.1                         | 
| Language                            |  {C++ / ROS (Wrapper V. 2.3.20}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hello Intel Team, we are having the following issue. 
Low rgb camera framerate issues. The camera node is publishing at ~4 Hz, even though it is set to 60. Tried it in realsense viewer, there it starts at normal rate but drops down to 4 Hz after a couple of seconds. Htop showed no significancies. All other streams seem to have stable framerate. 

We are also having this issue: (messenger-libusb.cpp:42) control_transfer returned error. At least this is also printed as a ROS_WARN every second.  But I am not sure if its correlated

**What we have tried so far:**
-Disabling autoexposure & enabling autoexposure priority. Changes the framerate to the normal rate but just for a couple of seconds, after that it drops again to 4 Hz.

- Changed orientation of USB-C Cable

- Tried a different USB-C Cable

- Changed resolution and framerate

- Switched all other sensors off

This is the launchfile we are using: 
`<launch>
  <arg name="serial_no"             default=""/>
  <arg name="json_file_path"        default=""/>
  <arg name="camera"                default="camera"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="align_depth"               default="false"/>
  <arg name="decimation_filter_magnitude"  default="3"/>
  <arg name="enable_infra1"		default="false"/>
  <arg name="enable_infra2"		default="false"/>
  <arg name="clip_distance"		default="5"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"         value="$(arg serial_no)"/>
      <arg name="json_file_path"    value="$(arg json_file_path)"/>
      <arg name="depth_width"       value="1024"/>
      <arg name="depth_height"      value="768"/>
      <arg name="depth_fps"         value="30"/>
      <arg name="color_width"       value="1280"/>
      <arg name="color_height"      value="720"/>
      <arg name="color_fps"         value="60"/>
      <arg name="enable_depth"      value="true"/>
      <arg name="enable_color"      value="true"/>
      <arg name="enable_infra1"     value="false"/>
      <arg name="enable_infra2"     value="false"/>
      <arg name="enable_fisheye"    value="false"/>
      <arg name="enable_gyro"       value="false"/>
      <arg name="enable_accel"      value="false"/>
      <arg name="enable_pointcloud" value="true"/>
      <arg name="enable_sync"       value="true"/>
      <arg name="tf_prefix"         value="$(arg camera)"/>
      <arg name="initial_reset"	    value="$(arg initial_reset)"/>
      <arg name="align_depth"	    value="$(arg align_depth)"/>
      <arg name="filters" default="pointcloud,decimation"/>
      <param name="decimation/filter_magnitude" value="$(arg decimation_filter_magnitude)"/>
    </include>
  </group>
</launch>`

libusb issue

brtmax, 2023-08-27:
`This is the launchfile we are using: 
`<launch>
  <arg name="serial_no"             default=""/>
  <arg name="json_file_path"        default=""/>
  <arg name="camera"                default="camera"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="align_depth"               default="false"/>
  <arg name="decimation_filter_magnitude"  default="3"/>
  <arg name="enable_infra1"		default="false"/>
  <arg name="enable_infra2"		default="false"/>
  <arg name="clip_distance"		default="5"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"         value="$(arg serial_no)"/>
      <arg name="json_file_path"    value="$(arg json_file_path)"/>
      <arg name="depth_width"       value="1024"/>
      <arg name="depth_height"      value="768"/>
      <arg name="depth_fps"         value="30"/>
      <arg name="color_width"       value="1280"/>
      <arg name="color_height"      value="720"/>
      <arg name="color_fps"         value="60"/>
      <arg name="enable_depth"      value="true"/>
      <arg name="enable_color"      value="true"/>
      <arg name="enable_infra1"     value="false"/>
      <arg name="enable_infra2"     value="false"/>
      <arg name="enable_fisheye"    value="false"/>
      <arg name="enable_gyro"       value="false"/>
      <arg name="enable_accel"      value="false"/>
      <arg name="enable_pointcloud" value="true"/>
      <arg name="enable_sync"       value="true"/>
      <arg name="tf_prefix"         value="$(arg camera)"/>
      <arg name="initial_reset"	    value="$(arg initial_reset)"/>
      <arg name="align_depth"	    value="$(arg align_depth)"/>
      <arg name="filters" default="pointcloud,decimation"/>
      <param name="decimation/filter_magnitude" value="$(arg decimation_filter_magnitude)"/>
    </include>
  </group>
</launch>``

dmipx, 2023-08-28:
What is your CPU usage?
Do you have enough resources?
Can you try realsense-viewer to capture color and see framerate is correct?
@SamerKhshiboun any idea?

brtmax, 2023-08-28:
Thank you for the answer!

CPU usage is normal, I also tested it on two devices, one with the Intel NUC and my personal laptop. Issue persists on both, and both should be able to handle the workload. We have a D435, and do not have the issue there and it should be the same RGB camera if I am informed correctly. 

When I start realsense-viewer the framerate is normal for a couple of seconds but drops down quickly. I tried playing around with the autoexposure features but turning them on/off did not seem to change anything

MartyG-RealSense, 2023-08-28:
Hi @brtmax  I note that you tried disabling auto-exposure and enabling auto-exposure priority.  The procedure for enforcing a constant FPS rate is actually the opposite.  Auto-exposure should be enabled and auto-exposure priority disabled.

You can also test whether pointcloud and / or alignment is causing your problem by setting them to false in your launch to see whether the FPS rate becomes correct.

brtmax, 2023-08-28:
Hello MartyG, thank you for the quick answer! I will try those actions and report back as soon as possible

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense