realsenseai/librealsense #12142: file:rs_rgbd_launch error&  Object Analytics (OA) unavailable

NET-Y, 2023-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   |                          5.13.0.55                        | 
| Operating System & Version |                  Linux (Ubuntu 18.04)             | 
| Kernel Version (Linux Only)    |                         5.4.0-150                        | 
| Platform                                 |                                PC                           |
| SDK Version                            |                             2.54.1                        | 
| Language                              |                                opencv                     | 
| Segment			        |                              Robot                         | 
  
### Issue Description
Hello:
We encountered several issues while trying to use a depth camera to locate an object:
1. We found that an error was reported when running the file:    rs_rgbd_launch :
    [ERROR] [1693205097.460124267]: Failed to load nodelet [/camera/realsense_ros_camera] of type 
   [realsense_ros_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin 
   realsense_ros_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that 
   the library actually exists.
   [ERROR] [1693205097.460205152]: The error before refreshing the cache was: Could not find library corresponding to plugin 
   realsense_ros_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that 
   the library actually exists.
   [FATAL] [1693205097.463399012]: Failed to load nodelet '/camera/realsense_ros_camera` of type ` 
   realsense_ros_camera/RealSenseNodeFactory` to manager `realsense_ros_camera_manager'
   [camera/realsense_ros_camera-3] process has died [pid 33838, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load 
   realsense_ros_camera/RealSenseNodeFactory realsense_ros_camera_manager __name:=realsense_ros_camera 
   __log:=/home/yang/.ros/log/6685f41c-456e-11ee-b3dd-000c29862474/camera-realsense_ros_camera-3.log].
   log file: /home/yang/.ros/log/6685f41c-456e-11ee-b3dd-000c29862474/camera-realsense_ros_camera-3*.log

[image]

2.We tried the Object Analytics provided by your official website but because of the software version that does not match we faild to run it. So we are wondering if there is a similar ROS wrapper that can be used on ubuntu 18.04 or 20.04.

3.As beginners,we would appreciate it if you can inspire us of how to use the camera to obtain the three-dimensional coordinates of a specific object

Thank you inadvance for your help!

MartyG-RealSense, 2023-08-28:
Hi @NET-Y  I would not recommend using the ROS1 version of Object Analytics because of its significant age.  It may be possible to get it working using ROS1 Kinetic, Ubuntu 16.04 and the **2.3.2** version of the RealSense ROS1 wrapper that supports Kinetic, but it is likely not worth doing so.

The 2.3.2 ROS1 wrapper works with ROS1 Kinetic / Melodic / Noetic and Ubuntu 16.04 / 18.04 / 20.04.  It should be used with librealsense version 2.50.0 or 2.51.1.  Because development of the ROS1 wrapper has ceased and only the ROS2 wrapper is updated, the ROS1 wrapper does not have support for librealsense 2.54.1.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

If you need to perform object detection in ROS1 with Ubuntu 18.04 then it would be preferable to use the Melodic branch of **darknet_ros**

https://github.com/leggedrobotics/darknet_ros/tree/melodic

If you require 3D coordinates then you could use the Melodic or Noetic branch of **darknet_ros_3d**

MELODIC
https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/tree/melodic

NOETIC
https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/tree/noetic

NET-Y, 2023-08-29:
> Hi @NET-Y I would not recommend using the ROS1 version of Object Analytics because of its significant age. It may be possible to get it working using ROS1 Kinetic, Ubuntu 16.04 and the **2.3.2** [...]

Thank you for your reply! We are trying darknet_ros_3d to reach our goal.  We have completed the configuration of Darknet Ros. But When configuring darknet_ros_3d，it reminds us of the lack of [gb_visual_detection_3d_msgs](https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d_msgs). We have no idea which folder to put this file in. And we fail to use CMake to compile the file. We would appreciate it if you could provide us with some suggestions.Thank you for your help!
[image]

MartyG-RealSense, 2023-08-29:
I do not know for certain where to clone **gb_visual_detection_3d_msgs** to, but the logical location would be the **src** catkin workplace, where darknet_ros and darknet_ros_3d are also installed.

NET-Y, 2023-08-30:
> I do not know for certain where to clone **gb_visual_detection_3d_msgs** to, but the logical location would be the **src** catkin workplace, where darknet_ros and darknet_ros_3d are also installed.

Thank you for your reply.We have solved the problem .So I guess this issue can be closed?

MartyG-RealSense, 2023-08-30:
It's great to hear that you succeeded!  As you have done so, I will close the case.  Thanks again!

closed 2023-08-30 by MartyG-RealSense