realsenseai/librealsense #12145: Failed to build RealSense SDK 2.0

19009yang, 2023-08-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Operating System & Version |   {Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  (4.15)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0 }                          | 

### Issue Description
During the process of running the installation guide on intelrealsense（https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide），When running the command ". /scripts/patch-realsense-ubuntu-lts.sh" in the librealsense directory, the following error message appears：

【Required package libusb-1.0-0-dev: - found
  Required package libssl-dev: - found
  Create patch workspace in ubuntu-xenial-hwe folder

  Fatal error: repository 'https://kernel.ubuntu.com/ubuntu/ubuntu-xenial.git/' not found
  Get Ubuntu LTS tags to local kernel source folder
  Fatal error: You need to specify a tag name】

MartyG-RealSense, 2023-08-29:
Hi @19009yang  As you are using Ubuntu 16.04, the simpler installation method at the link below for installing librealsense from source code may work for you.  It is not necessary to use a kernel patch script with this method.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2023-09-05:
Hi @19009yang  Do you require further assistance with this case, please?  Thanks!

19009yang, 2023-09-06:
> 你好@19009yang 请问您对此案需要进一步的帮助吗？谢谢！
I’ve solved it. Thank you!

MartyG-RealSense, 2023-09-06:
You are very welcome.  Thanks very much for the update!  As you have achieved a solution, I will close this case.  Thanks again!

非常歡迎您。 非常感謝您的更新！ 當您已經找到解決方案時，我將關閉此案例。 再次感謝！

closed 2023-09-06 by MartyG-RealSense