realsenseai/librealsense #12147: How to get point cloud to display in Unity without pressing the play button?

willyum00, 2023-08-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Win 11  | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            | unity | 
| Segment			| AR | 

### Issue Description
Hi Marty,

As the title states, how do I get a point cloud to display in Unity without having to press the play button? I am trying to do something similar to this:
[image]
from this video: https://youtu.be/SrLhF7sMfyk?si=9syqYed3tiKUO7ZC

This is how my unity looks right now:
[image]

it's difficult to get the correct configuration without knowing exactly where the point cloud is in space. 

Thanks!

willyum00, 2023-08-30:
quick update: 
I am now able to display a point cloud, but it looks messed up, any idea why? Thanks!

[image]

willyum00, 2023-08-30:
After further tooling, I was able to get it working alright by adjust the size of my frame. However, I am still experiencing an issue where only certain parts are completely captured. ie only objects within a certain distance appear, while everything else is black. However, this could be an issue with the Looking Glass, not realsense.

MartyG-RealSense, 2023-08-30:
Hi @willyum00  It's great to hear that you were made significant progress!

The Looking Glass video that you quoted was filmed with a RealSense **D435i** camera model.  this camera is equipped with a projector that casts a dot-pattern projection onto objects in the real-world scene.  This improves depth image quality by enabling the camera to analyze the dots on object surfaces for depth information.

The D405 model that you are using is not equipped with a projector though, so surfaces that are plain or smooth will be more difficult for the D405 camera to analyze for depth detail.  Increasing the lighting level in the real-world scene in the area that the camera is observing may improve the image, as 400 Series cameras can alternatively use ambient light in a scene for depth analysis instead of the dot pattern.

Also, the D405 is designed to produce high quality, high accuracy images at close range.  Its ideal depth sensing range is 7 cm to 50 cm (0.5 meters).  The D435i model can depth-sense up to 10 meters.  So with the D405, the background will be mostly black because its depth sensing range does not extend far.

closed 2023-09-05 by willyum00