realsenseai/librealsense #12150: libusb cmake not configuring correctly

Kikadass, 2023-08-30:
I am trying to get the repo through FetchContent and compiling the code.
However, I did not install libusb-1.0-0-dev, and I see in the cmake configuration that this project is supposed to download it itself through ExternalProject found in CMake/external_libusb.cmake (https://github.com/IntelRealSense/librealsense/blob/master/CMake/external_libusb.cmake)

The problem is that libusb/core.c needs config.h, which I believe to be generated through autoconf, but this repo is not using autoconf to generate those files. Therefor, there a compiler issue as found in issue https://github.com/IntelRealSense/librealsense/issues/4469

I fixed it by downloading libusb-1.0-0-dev, however, I still think this is a bug, as the code to download libusb exists inside this CMake project, but is not working. 
Therefor, I think the external_libusb.cmake should be fixed.

I appreciate your hard work.

MartyG-RealSense, 2023-08-30:
Hi @Kikadass  The official instructions for installing the librealsense SDK from source code can be found at the link below. 

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

A command for manually installing **libusb-1.0-0-dev** is included in the list of instructions.

`sudo apt-get install libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev cmake`

Kikadass, 2023-08-31:
As I mentioned, I know that installing libusb manually will fix it, but in the CMake code there is a mechanism to try to install it if it its missing. 

That mechanism is broken, therefor I raised an issue/bug. I would recommend fixing it or removing it completely.

MartyG-RealSense, 2023-08-31:
I will pass your suggestion on to the development team.

MartyG-RealSense, 2023-09-13:
Case closed after feedback passed on to development team.

closed 2023-09-13 by MartyG-RealSense

Ryanf55, 2024-09-07:
This affects the last supported version for L515, at version v2.51.1.