realsenseai/librealsense #12156: IR image totally underexposed. No ambient light. Laser Power 95.

MarcoWyss, 2023-09-01:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0264135|
|**Firmware Version**|1.5.8.1|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003ac793907|
|**Firmware Update Id**|0003ac793907|

Please provide a description of the problem

MartyG-RealSense, 2023-09-01:
Hi @MarcoWyss  Are you using the **No Ambient Light** camera configuration preset?  This preset is intended for use in rooms with no natural light coming in through windows.  The L515 camera model can experience interference from IR light sources such as sunlight.

If there is some natural light present in the room then the **Low Ambient Light** preset will be suitable.  

**Short Range** is a good preset to use in general instead of the default preset **Maximum Range**.

MarcoWyss, 2023-09-04:
Hi Marty

Thank you very much for your quick reply.
I tested it also with excluded ambient light (very dark room, no artificial lighting, no sunlight). In all the modes (Short, Maximum, Ambient light, low ambient light), but no big improvement.
The camera worked well in the beginning. But then suddenly, the 3D images were very noisy and the IR so dark, that nothing can be recognized.

Best regards,
Marco

MartyG-RealSense, 2023-09-04:
If you are using the RealSense Viewer with your L515, please check if an option called 'Alternate IR' has been enabled (highlighted blue).  It considerably darkens the IR image and remains enabled even when the Viewer is closed and re-launched until it is manually disabled.

[image]

MarcoWyss, 2023-09-04:
Hi Marty

Thank you very much for your answer. I already tried to switch “Alternate IR” on and off.
Yes, when it is on, the IR image is brighter, but practically no contrast, nothing recognizable in the image. That was different in the beginning.
When it is off, the IR image is very dark, also nothing recognizable.

Best regards,
Marco

[image]<https://user-images.githubusercontent.com/41145062/265384661-61433e9e-770c-4d24-977f-92b544a601fd.png>

MartyG-RealSense, 2023-09-04:
Let's try running the 'rs-multicam' example program to test the depth and IR outside of the Viewer.  If you have installed the full RealSense SDK on Windows with the 'Intel. Realsense.SDK-WIN10' installer program then you can find and launch rs-multicam.exe at the location on your computer below.

C: > Program Files (x86) > Intel RealSense SDK 2.0 > tools

A quick way to find this folder is to right-click on the Viewer's launch shortcut icon on the Windows desktop and select the menu option 'Open file location'.

[image]

If the infrared is still very dark in rs-multicam then there may be an internal hardware malfuction in your L515.  If you purchased it relatively recently then you can contact the retailer that you purchased it from to enquire about returning the camera,

MarcoWyss, 2023-09-04:
Hi Marty

Here are the images.
First, with the Intel RealSense Viewer (newest version): Artificial light is switched off, no direct sunlight but moderate natural light through the windows (see RGB stream):
***@***.***
And here the very noisy 3D image:
***@***.***

The same conditions, but now with the rs-multicam.exe:
***@***.***

[image]<https://user-images.githubusercontent.com/41145062/265420098-335b49df-6427-40a1-b6e5-8de679854130.png>

MarcoWyss, 2023-09-04:
Hi Marty

Here are the images.
First, with the Intel RealSense Viewer (newest version): Artificial light is switched off, no direct sunlight but moderate natural light through the windows (see RGB stream):
[image]
And here the very noisy 3D image:
[image]
The same conditions, but now with the rs-multicam.exe:
[image]

MartyG-RealSense, 2023-09-04:
The infrared image should sharpen and become less blurry if Laser Power is maximized to '100'.  You can also have some manual control over the IR brightness by adjusting 'Receiver Gain'.  Higher gain = darker IR, lower gain = lighter IR.

MartyG-RealSense, 2023-09-06:
Below is a picture of a bathroom in the dark with an L515 using the No Ambient Light preset.

[image]

The L515 camera model does not have a factory reset function, unfortunately.

Is there any difference if you unplug the micro-sized end of the USB cable from the base of the camera, reverse the connector's orientation and plug it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).

MarcoWyss, 2023-09-06:
Here is a comparable image. Made in total darkness:
[image]

MartyG-RealSense, 2023-09-06:
There are no problems in your settings or the camera temperatures.  It looks as though the camera's laser is having difficulty illuminating the scene.  Since the Laser Power is maximized, that makes me think that it's a camera hardware problem.

MarcoWyss, 2023-09-07:
Hi Marty, 
yes I am thinking also more and more, that it's most problably a defect of the IR camera. As we bought the L515 more than 2 years ago I don't think that we still have any warranty. Are spare parts available, i.e., the IR camera?

MartyG-RealSense, 2023-09-07:
The L515 camera is not a repairable device and spare parts for it are not available.  And yes, the warranty period for RealSense cameras is 1 year.

Does the infrared image become more readable if you use **Low Gain** as the Digital Gain settng instead of the default High Gain?  Low Gain should darken the IR image.

[image]

MarcoWyss, 2023-09-07:
High gain:
[image]
Low gain:
[image]

Switching of "Alternate IR" makes the situation even worse.

MartyG-RealSense, 2023-09-07:
High Gain may be the best image that you can achieve with the issue on your particular camera, then.

MarcoWyss, 2023-09-07:
Ok, thank you very much for your fast and professional support!

MartyG-RealSense, 2023-09-07:
You are very welcome!

MartyG-RealSense, 2023-09-13:
Hi @MarcoWyss  Do you require further assistance with this case, please?  Thanks!

MarcoWyss, 2023-09-13:
Hi Marty

No, but again thank you very much for your support.
And I have to say that it is disappointing that the L515 is a non-repairable device. It could have been built to be repairable for environmental reasons.

Best regards,
Marco

MartyG-RealSense, 2023-09-13:
Thanks very much for your patient understanding.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-09-13 by MartyG-RealSense