realsenseai/librealsense #12157: D435i not working in Windows 11

quebeh, 2023-09-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Windows 11  | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.1 | 

### D435i not working in Windows 11
Hello, I have an Intel RealSense D435i and a Windows 11 laptop.
According to the releases page, the SDK should work with Windows 11..
So I installed the SDK with the `Intel.RealSense.SDK-WIN10-2.54.1.5216.exe` in the releases page.

When I open the Intel RealSense Viewer application, it is able to successfully recognize the device.
But when I enable the stereo camera, the display is completely white with a text that says "No frames received". The same happened with the color camera.
But if I tweak the FPS to 60, it successfully received frames through the RealSense.. but It's very slow and not quite 60 FPS.
For some reason, the motion frames works, even though the response time is slow.
I even tried to read it through python, and it didn't work either, `Frame didn't arrive within 5000`.

I read other issues.. but none seems to solve my issue.
I tried on my friend's laptop and it worked properly.. (with Windows 10)
And it also worked with a Jetson Nano running Ubuntu

Please do tell me if there is any solution.
Thanks in advance

quebeh, 2023-09-01:
Yes, I tried turning on the stereo and RGB cameras independently and it still does not receive any frames.

MartyG-RealSense, 2023-09-01:
Thank you.  Is the camera attached to the USB port of the computer or are you using a USB hub?

quebeh, 2023-09-01:
It is attached directly to the USB port of the laptop

MartyG-RealSense, 2023-09-01:
If it works on your friend's Windows laptop and a Nano with Ubuntu then it may be an issue with the USB ports on your laptop, in particular their **USB controller**.  The brand of USB controller is chosen by the computer's manufacturer and some brands may work better than others with RealSense cameras.

A way to confirm whether it is the USB controller on your laptop would be to attach the camera to a USB hub if possible to do so, as a hub uses its own brand of USB controller separate from the built-in ports of the computer.

quebeh, 2023-09-01:
Oh, it works with a USB hub. 
FYI the laptop brand is an MSI laptop and the USB hub brand is Toshiba. 

Thank you very much.

closed 2023-09-01 by quebeh

MartyG-RealSense, 2023-09-01:
You are very welcome.  It's great to hear that a hub worked for you  :)