realsenseai/librealsense #12158: librealsense2-dkms 1.3.22-0ubuntu1: librealsense2-dkms kernel module failed to build - 5.15 iotg kernel

ionutnechita-intel, 2023-09-01:
Error Output:

```
apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-5.15.0-1026-intel-iotg linux-image-5.15.0-1026-intel-iotg linux-intel-iotg-headers-5.15.0-1026 linux-modules-5.15.0-1026-intel-iotg
  linux-modules-extra-5.15.0-1026-intel-iotg
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  librealsense2-dkms
0 upgraded, 1 newly installed, 0 to remove and 14 not upgraded.
Need to get 0 B/4.445 kB of archives.
After this operation, 23,7 MB of additional disk space will be used.
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 266035 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.22-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.22-0ubuntu1) ...
Setting up librealsense2-dkms (1.3.22-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.22 DKMS files...
Building for 5.15.0-1038-intel-iotg
Building initial module for 5.15.0-1038-intel-iotg
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Application of patch 92-realsense-hid-5.15.patch failed.
Check /var/lib/dkms/librealsense2-dkms/1.3.22/build/ for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 6
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

File: librealsense2-dkms.0.crash
```
cat /var/crash/librealsense2-dkms.0.crash
ProblemType: Package
DKMSKernelVersion: 5.15.0-1038-intel-iotg
Date: Wed Aug 30 19:28:57 2023
Package: librealsense2-dkms 1.3.22-0ubuntu1
PackageVersion: 1.3.22-0ubuntu1
SourcePackage: librealsense2
Title: librealsense2-dkms 1.3.22-0ubuntu1: librealsense2-dkms kernel module failed to build
```
OS file - os-release:
```
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
```

Kernel Version:
```
uname -a
Linux my-machine 5.15.0-1038-intel-iotg #43-Ubuntu SMP Wed Aug 16 21:04:08 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

OS: Ubuntu IOT Desktop 22.04 LTS
Camera: Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i

MartyG-RealSense, 2023-09-01:
Hi @ionutnechita-intel  As far as I am aware, **Ubuntu IOT Desktop** has not been previously reported on this support forum to have been used with RealSense cameras.

Do you have the option of building the librealsense SDK from source code instead of installing with packages, please?  If the SDK is built from source code with CMake and the flag **-DFORCE_RSUSB_BACKEND=TRUE** is included in the CMake build instruction then the SDK will not be dependent on Linux versions or kernel versions and not require a kernel patch.  This makes an RSUSB build of the SDK good for more unusual hardware or OS specifications than the typical consumer PC.

MartyG-RealSense, 2023-09-09:
Hi @ionutnechita-intel  Do you require further assistance with this case, please?  Thanks!

ionutnechita-intel, 2023-09-11:
Hi @MartyG-RealSense ,
For the installation from the sources, I will try to do all the installation steps myself.

But for Ubuntu IOT, can we have an updated patch for this type of kernel in the next version of librealsense?

MartyG-RealSense, 2023-09-11:
I will ask one of my Intel RealSense colleagues who specializes in kernels.

@dmipx  Could you provide advice about the comment above please regarding kernel 5.15 support for Ubuntu IOT?

dmipx, 2023-09-13:
Hi @ionutnechita-intel
Recently we submitted bugfix for hid and latest kernels doesn't need hid patch.
We have fix for it but new dkms was not released yet.
You can build modules manually, from 'development' branch by running
./scripts/patch-realsense-ubuntu-lts-hwe.sh
https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-lts-hwe.sh
If you have internal access to artifactory, you can use [this link](https://ubit-artifactory-il.intel.com/artifactory/realsense_debian-il-local/pool/jammy/main/librealsense2-dkms_1.3.24-0ubuntu1_all.deb) to get updated dkms for jammy

ionutnechita-intel, 2023-09-22:
When will we have the new version, 1.3.24?

MartyG-RealSense, 2023-09-28:
Hi @dmipx  Could you assist @ionutnechita-intel with the question in the above comment please?  Thanks!

dmipx, 2023-10-01:
Hi. I'm checking with team when DKMS will pe promoted.

dmipx, 2023-10-04:
We promoted DKMS 1.3.24, you can try now.

ionutnechita-intel, 2023-10-04:
> We promoted DKMS 1.3.24, you can try now.

Ok. Will try to testing with this version.

ionutnechita-intel, 2023-10-06:
Hi @dmipx,

I tested today and this version working correctly on Ubuntu 22.04 IOT. Kernel and modules match now.

// ## modinfo videodev
filename:       /lib/modules/5.15.0-1042-intel-iotg/updates/dkms/videodev.ko
version:        realsense-1.3.24
alias:          char-major-81-*
license:        GPL
description:    Video4Linux2 core driver
author:         Alan Cox, Mauro Carvalho Chehab <mchehab@kernel.org>, Bill Dirks, Justin Schoeman, Gerd Knorr
srcversion:     9A4FB8163941EA975F2CF8A
depends:        mc
retpoline:      Y
name:           videodev
vermagic:       5.15.0-1042-intel-iotg SMP mod_unload modversions
sig_id:         PKCS#7
signer:         glaic3n170 Secure Boot Module Signature key
sig_key:        3D:B9:3E:D6:3E:87:9C:15:88:C4:79:0C:4E:9B:A8:CA:85:09:F6:0D
sig_hashalgo:   sha512
signature:      B3:92:74:60:F9:46:3E:D7:1C:9E:C4:76:9D:B9:6A:27:F7:4C:55:31:
                F4:B0:CF:83:40:0E:A5:7B:AE:3C:8C:48:3B:AE:8D:B9:D9:81:42:26:
                5E:6B:B8:28:D1:5B:EC:4D:21:08:87:C8:F5:E7:06:09:26:CB:99:7D:
                0F:EA:9B:C3:A8:1E:15:40:18:8C:1F:DD:23:B8:C2:81:B1:D2:54:DF:
                FB:2E:A6:B0:E3:D6:8B:C0:1C:A6:A8:27:8B:17:A0:50:51:FE:30:A1:
                29:7A:FE:08:ED:75:B2:5C:7D:08:F0:15:A7:47:C6:6F:F8:42:C4:B0:
                13:CE:BC:03:C3:AB:F4:0E:E1:78:10:F1:B5:4C:18:9B:C0:9E:5A:9F:
                9C:34:B0:34:C0:9C:62:F2:70:A5:50:9D:8B:F8:01:2F:F1:97:73:6A:
                DD:86:98:1F:99:C9:84:C0:86:4E:78:B0:97:C9:E7:99:7D:6C:EB:9C:
                ED:57:68:9F:DB:20:7A:F2:75:C5:58:C6:F9:50:39:4B:01:05:CE:1E:
                1E:B4:D4:FA:76:27:72:6C:F6:03:90:33:3C:77:B8:8E:47:13:82:B3:
                C3:D6:78:09:30:FB:AA:17:4A:C1:22:D5:ED:76:60:1C:A3:79:7D:24:
                84:7A:21:1A:81:12:F2:5E:19:35:9D:F7:44:21:72:8B

Thank you for all support.

Ticket is verified.

Please close this ticket.

MartyG-RealSense, 2023-10-11:
Hi @ionutnechita-intel  It's great to hear that you achieved a solution.  As you have done so, i will close this ticket as you suggested.  Thanks again!

closed 2023-10-11 by MartyG-RealSense