realsenseai/librealsense #12159: Unreal Engin 5 Support

ranjitkathiriya544, 2023-09-01:
unreal engin 5 wapper for realsense.

I am looking for a GitHub repo on that.

https://github.com/IntelRealSense/librealsense/issues/11893

MartyG-RealSense, 2023-09-01:
Hi @ranjitkathiriya544  I see that you have found the RealSense interface for Unreal Engine 5 that a RealSense user created.

The creator of the interface did not provide a link to download their project, so discussing it with them at https://github.com/IntelRealSense/librealsense/issues/11893 will be the best course of action.

ffreality, 2023-09-01:
Hi, I am the developer of UE5 realsense plugin. We use it to give QR detection, object tracking features to Oculus Quest devices for our commerical projects and it is important for us as a trump card while competing with our competitors.

So, we don't consider to publish it as free open source.

But I can add developers as contributor if they can help us to improve it.

MartyG-RealSense, 2023-09-07:
Hi @ranjitkathiriya544  Do you have an update that you can provide for us or @ffreality please?  Thanks!

ffreality, 2023-09-07:
@MartyG-RealSense I think also Intel doesn't interest about developing official UE5 wrapper with crossplatform support.

MartyG-RealSense, 2023-09-07:
@ffreality  There are no plans to officially support Unreal Engine 5, though contributions of support interfaces from the RealSense community are welcome.

ffreality, 2023-09-07:
> @ffreality There are no plans to officially support Unreal Engine 5, though contributions of support interfaces from the RealSense community are welcome.

Problem is this :D everyone welcomes it but no one wants to help it. this is the problem of open source ecosystem.

MartyG-RealSense, 2023-09-13:
Hi @ranjitkathiriya544  Bearing in mind the advice of @ffreality above, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2023-09-19:
Case closed due to no further comments received.

Markyunfei, 2025-03-28:
我希望 RealSense 官方能够出一个关于虚幻五相关的使用插件，这对我和我的团队来说很重要。

ffreality, 2025-03-28:
> 我希望 RealSense 官方能够出一个关于虚幻五相关的使用插件，这对我和我的团队来说很重要。

https://github.com/Frozen-Projects/FF_RealSense

WhittSellers, 2025-05-06:
Hey @ffreality thank you for sharing your plugin! I'm currently trying to get it enabled in a 5.4.4 project targeting windows and don't need the qr reading. In looking through the readme instructions, I wasn't able to find any ZXing bp in the content folder, and removing the qr plugin dependency from the uplugin file results in needing to rebuild the plugin, and it looks like the qr plugin url in the readme is 404. Is there something I am missing about enabling the plugin? Thank you for any insight!

ffreality, 2025-05-07:
> Hey [@ffreality](https://github.com/ffreality) thank you for sharing your plugin! I'm currently trying to get it enabled in a 5.4.4 project targeting windows and don't need the qr reading. In [...]

I modified didn't use ZXing source code because building it for Android was very hard. So, I modified marketplace plugin to pass bgra buffer directly to zxing functions in FRunnableThread. I can't share Zxing plugin because it will be piracy.

For realsense perspective, I don't offer service about it and I abandoned it because I needed to attach it Quest 2 / Pro to give QR reader feature to VR headset. Right now we have Camera2 API access with Quest 3.

And you talked like rebuilding plugin is a big stuff like rebuilding whole engine. You have to rebuild it nevertheless because I didn't gave pre built binaries. It is just source code. So, don't be lazy. Install Visual Studio, fork my repo, remove any Zxing related stuff and press F5 from your keyboard. That's all. You won't do any coding at all.

WhittSellers, 2025-05-07:
Ah ok I see about Zxing, and totally makes sense about abandoning realsense. I appreciate the response at all! I haven't actually built a plugin before, but yeah that doesn't sound too bad, I'll give it a shot. Thank you!