realsenseai/librealsense #12176: How to apply decimation filter on D405

LamiumAmplexicaule, 2023-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 22.04 Jammy  | 
| Kernel Version (Linux Only)    | 6.2.0-32-generic  | 
| Platform                                 | PC |
| SDK Version                            | 2.54.1 | 
| Language                            | Python | 
| Segment			| Others | 

### Issue Description

D435i can apply decimation filter to color frames with realsense-viewer, but not with the [sample code](#Sample-Code). D405 cannot apply the decimation filter in either realsense-viewer or the [sample code](#Sample-Code).

How can I apply decimation filter to color frame and depth frame with D405?

#### Sample Code

```python
import pyrealsense2 as rs

def main():
    ctx = rs.context()
    pipeline = rs.pipeline(ctx)

    config = rs.config()

    pipeline_wrapper = rs.pipeline_wrapper(pipeline)
    pipeline_profile = config.resolve(pipeline_wrapper)
    device = pipeline_profile.get_device()
    device_name = str(device.get_info(rs.camera_info.name))

    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    decimation = rs.decimation_filter(2)

    print(f"Device name: {device_name}")

    _profile = pipeline.start(config)
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        depth_frame = decimation.process(depth_frame).as_depth_frame()
        color_frame = decimation.process(color_frame).as_video_frame()
        print(f"Depth frame: ({depth_frame.get_width()}, {depth_frame.get_height()})")
        print(f"Color frame: ({color_frame.get_width()}, {color_frame.get_height()})")

if __name__ == '__main__':
    main()

```

#### Sample Result

```bash
(venv) henbit@i7-9700 /tmp> python down_sampling.py
Device name: Intel RealSense D405
Depth frame: (320, 240)
Color frame: (640, 480)
Depth frame: (320, 240)
Color frame: (640, 480)
Depth frame: (320, 240)
Color frame: (640, 480)
Depth frame: (320, 240)
Color frame: (640, 480)
................
```

MartyG-RealSense, 2023-09-11:
Hi @LamiumAmplexicaule  The D405 camera does not have an RGB sensor to apply the Decimation filter to, as the RGB image is provided by the **depth** sensor on D405 and then processed with an Image Signal Processor (ISP) chip inside the camera to produce the final RGB image.  The Decimation filter does downsample the resolution of the depth image on D405 though.

The Decimation filter divides the resolution by whatever value is set for the filter.  The default value for the filter is '2', which is what you have set in your script.  So if depth is set to 640x480 resolution then a Decimation value of '2' downsamples the depth resolution to 320x240, as shown in your frame size print-out.

For example, the 848x480 depth image below from the Viewer with frame information overlaid on the image demonstrates that if the filter is set to '3' then the image **display size** is divided by 3 (284x160).

[image]

Changing the Decimation filter value on D405 does not affect the RGB color image, only the depth image.

LamiumAmplexicaule, 2023-09-11:
@MartyG-RealSense 

Thank you for your response.
So, if I want to decrease the resolution of the D405 color image to match the resolution of the depth frame, should I use software like OpenCV to decrease the resolution of the color frame?

### Sample Code

```python
cv2.resize(np.asarray(color_frame.get_data()), (320, 240), interpolation=cv2.INTER_AREA)
```

MartyG-RealSense, 2023-09-11:
The code that you quoted would be one method of custom configuring the RGB display resolution, yes.

The simplest way to match the resolutions on D405 though may be to just choose a low depth resolution.  Because RGB color comes from the depth sensor, RGB will automatically use the same resolution.  For example, choosing 424x240 will set both depth and color to that resolution.

[image]

Another way to match different resolutions is to use depth to color alignment (align_to), as the RealSense SDK's **align processing block** automatically adjusts for resolution differences between depth and color streams when calculating an aligned image.

LamiumAmplexicaule, 2023-09-12:
It was very helpful. Thank you.

closed 2023-09-12 by LamiumAmplexicaule

MartyG-RealSense, 2023-09-12:
You are very welcome.  :)