realsenseai/librealsense #12180: Cannot flash firmware via GMSL

nathanieltagg, 2023-09-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R457 (GMSL connectors) | 
| Firmware Version                   | 5.13.1.53 | 
| Operating System & Version |  Ubuntu 20.04 (Jetpack 5.0.2)  | 
| Kernel Version (Linux Only)    |  5.10.104                                      | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            | R/254                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                           | 

### Issue Description
Flashing firmware fails both from realsense-viewer, and using the rs-fw-update tool.

```$ rs-fw-update -s 243122300563 -f Signed_Image_UVC_5_15_0_2.bin

Search for device with serial number: 243122300563

Updating device:
Name: Intel RealSense D457, serial number: 243122300563, update serial number: 250343111884, firmware version: 5.13.1.53, USB type: unknown
 11/09 16:43:10,317 WARNING [281473866676480] (ds-device-common.cpp:93) Timeout waiting for device disconnect after DFU command!

Failed to locate a device in FW update mode
```

Tried again with sudo on another camera (may be related to above failure, did not reboot in between):
```$ sudo rs-fw-update -s 242422305103 -f Signed_Image_UVC_5_15_0_2.bin

Search for device with serial number: 242422305103
 11/09 16:48:19,493 WARNING [281473133332736] (rs.cpp:312) null pointer passed for argument "device"
RealSense error calling rs2_create_device(info_list:0xaaaaede0fc50, index:0):
    hwmon command 0x10( 0 0 0 0 ) failed (response 0= Success)```

I also tried the cat to /dev/df4xx-dfu504 without success.

MartyG-RealSense, 2023-09-11:
Hi @nathanieltagg  The ability to use a MIPI connection in rs-fw-update for D457 was added to the **development** branch of the RealSense SDK in August 2023 and so should be in the next SDK release after the current 2.54.1.

https://github.com/IntelRealSense/librealsense/pull/12104

The source code of the development branch of librealsense, which becomes the next master release of the SDK when published, can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/development

In the meantime you could switch the camera to **USB** mode using the mode switch under the small removable cap on top of the camera to perform a firmware update over USB and then change the mode switch back to 'MIPI' afterwards.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

nathanieltagg, 2023-09-12:
Running the development build did not work; it gave reasonable startup messages, and then just stalled for many minutes with no activity.

We spent the morning unmounting cameras, switching to USB, pushing new firmware, and remounting.   Eventually we will need a solution to this, though; our product requires us to update over-the-air.

MartyG-RealSense, 2023-09-13:
Thanks very much for the update about the temporary solution of switching to USB to update the firmware.  You should not need to update the firmware again until the next SDK release, so hopefully the updated rs-fw-update in that release will enable you to update firmware without unmounting the cameras.  The alternative would be to remain with SDK 2.54.1 for a longer period so that a firmware update is not required.

MartyG-RealSense, 2023-09-28:
Hi @nathanieltagg  Support for D457 has been added to the rs-fw-update tool in the latest 2.54.2 librealsense release.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

closed 2023-09-29 by nathanieltagg