realsenseai/librealsense #12182: how can I find the corresponding coordinates of these RGB values in the point cloud? If this is not possible with RGB values, how can I find them with RGBD values?

Fc6868, 2023-09-12:
Hi  I am a new user of D405 depth camera. In my project, I need to detect an object and find its 3D coordinates in the point cloud. I detect the object on the RGB image and get the RGB values of the object. At the same time, I record the object in the point cloud with the D405 camera. I read the coordinates in the point cloud with open3d.
 how can I find the corresponding coordinates of these RGB values in the point cloud? If this is not possible with RGB values, how can I find them with RGBD values?

MartyG-RealSense, 2023-09-12:
Hi @Fc6868  If your pointcloud is an RGBD **textured pointcloud** where the RGB color data is mapped onto the depth pointcloud to produce an aligned image then the RGB coordinate of a particular point should be the same as the depth coordinate.

Fc6868, 2023-09-12:
Thanks for your response. What I need to understand from this message is that I can find the correct coordinates by using the D(depth) and Z(z-axis) equality condition to find the coordinate values in the point cloud. If this statement is correct, you can close the topic. have a good day.

MartyG-RealSense, 2023-09-12:
The depth value and z-axis value would be the same thing, since depth = z-distance.

closed 2023-09-12 by Fc6868