realsenseai/librealsense #12183: Wrong depth value after setting depth units 0.0001 in python

GuoCHance, 2023-09-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.14.00.00| 
| Operating System & Version |   WIN11  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy }                          | 
| Language                            |  python                         | 
| Segment			|  {AR }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

In Realsnese viewer, I exported pointcloud by setting depth units 0.001 and 0.0001. The depth value of the left bottom points is about 0.475m(one label is 0.001 and the other is 0.0001 depth units)
[image]

And then I tried to check the vertices value in python.(I only change the depth units in Realsense viewer)
code:
if __name__ == "__main__":
    #rs setup
    pipeline = rs.pipeline()
    config = rs.config()
    width = 1280
    height = 720
    config.enable_stream(rs.stream.depth, width, height, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, width, height, rs.format.bgr8, 30)
    profile = pipeline.start(config)
    detector = dlib.get_frontal_face_detector()
    Run = True
    QR_6dof = ''
    Head_6dof = ''
    img_idx = 0

    #filter declare
    threshold = rs.threshold_filter()
    threshold.set_option(rs.option.max_distance, 0.6)
    temp_filter = rs.temporal_filter()
    spat_filter = rs.spatial_filter()
    hole_filling_filter = rs.hole_filling_filter()
    dec_filter = rs.decimation_filter()
    dec_filter.set_option(rs.option.filter_magnitude,1)

    # align process
    align_to = rs.stream.depth
    align = rs.align(align_to)

    #Rs show image
    while Run:

        frames = pipeline.wait_for_frames()
        frames = align.process(frames)
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        #filter process
        depth_frame_filtered = threshold.process(depth_frame)
        #depth_frame_filtered = dec_filter.process(depth_frame_filtered)
        depth_frame_filtered = spat_filter.process(depth_frame_filtered)
        depth_frame_filtered = temp_filter.process(depth_frame_filtered)
        # depth_frame_filtered = rs.hdr_merge().process(depth_frame_filtered)
        # depth_frame_filtered = hole_filling_filter.process(depth_frame_filtered)

        #cv2 process image, show detected qrcode & face
        depth_image = np.asanyarray(depth_frame_filtered.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        faces = detector(color_image)
        qrcode_points = qr_detect(color_image)
        if qrcode_points[0]:
            qrcode_coord = np.empty((0, 2), dtype=np.float64)
            for point in qrcode_points[1:]:
                for i in range(4):
                    x, y = point[0][i]
                    qrcode_coord = np.vstack((qrcode_coord, (x, y)))
                    cv2.circle(color_image, (int(x), int(y)), 2, (0, 0, 255), -1)
        key = cv2.waitKey(1)
        if key == ord('t') or key == ord('T'):
            if qrcode_points[0]:
                pc = rs.pointcloud()
                pc.map_to(color_frame)
                points = pc.calculate(depth_frame_filtered)
                points.export_to_ply("1.ply", color_frame)
                vertices = np.reshape(np.asanyarray(points.get_vertices()), (height, width))
                print("QRCode Coordinates are: ")
                if qrcode_points[0]:
                    qrcode_corner = np.empty((0, 3), dtype=np.float64)
                    qrcode_v = np.empty((0,3),dtype=np.float64)
                    for i in range(4):
                        x, y = qrcode_coord[i]
                        print("x,y before:{},{}".format(x,y))
                        x = round(x)
                        y = round(y)
                        qrcode_corner = np.vstack(
                            (qrcode_corner, (vertices[y][x][0], -vertices[y][x][1], vertices[y][x][2])))
                        print("QRCode vertices " + str(i) + " is: " + str((x, y)) + " " + str(
                            (vertices[y][x][0], -vertices[y][x][1], vertices[y][x][2])))

The result is :
Depth units 0.001-> QRCode vertices 0 is: (569, 609) (-0.03373104, -0.124135286, 0.47500002)
Depth units 0.0001 -> QRCode vertices 0 is: (569, 609) (-0.03358191, -0.12358646, 0.47289997)

the x,y value is a little bit different because of the RGB detect error.But it should not lead to 2mm depth value difference.

How to correct the Depth value in python when setting Depth units 0.0001?
Thanks

GuoCHance, 2023-09-12:
Hi, MartyG!
Sorry for confusing discription.
Actually, my question is why i got different depth value using depth units 0.0001 (0.473m in python compared to 0.475m in Realsense Viewer). But i got same depth value using depth units 0.001(0.475m in python compared to 0.475m in Realsense Viewer.)

And yes, i will improve my filters list. Thanks for suggestion.!

Besides, I have tried to set the depth units in python using :
profile = pipeline.start(config)
depth_units = 0.0001
profile.get_device().first_depth_sensor().set_option(rs.option.depth_units,depth_units)

The result is same as  i change the depth units in Realsense Viewer->Setero Module->Controls

I used the align_to because I used to get the pointclouds in Color Camera coordinates.I changed the code align to depth because I want to check if the pointcloud x,y same to realsense viewer.

And actually I didn't used the decimation filter. I also close it in realsense viewer.

I have to set the depth unit 0.0001 because the depth image noise is so bad in depth units 0.001 that made other parts got bad result. depth unit 0.0001 had a better performance.

MartyG-RealSense, 2023-09-12:
If the problem is that the depth is fluctuating wildly then instead of setting the 0.0001 scale, setting the **Filter Smooth Alpha** option of the Temporal filter to '0.1' instead of its default of '0.4' may help.  Doing so can greatly help to stabilize fluctuating depth.  See https://github.com/IntelRealSense/librealsense/issues/10963#issuecomment-1268093845

GuoCHance, 2023-09-12:
I tried to change the option and I don't think the filter smooth alpha work well.
[image]
you can see there is bad noise on the mesh when setting the 0.001 scale and alpha 0.1
[image]
setting 0.0001 scale got better performance.

GuoCHance, 2023-09-13:
Have already read the #4629 comment, but unfortunately no help.

"I try to set the Marker in different place. I got different values in python scripts and Realsense Viewer export(using depth unit 0.001 or 0.0001). I think there maybe something wrong with my depth frame process in python?

Or, how should i improve my depth quality and stability in python? The depth value should be same whether I use depth units 0.001 or 0.0001. I want to get the depth value same in Realsense Viewer."

Mentioned before, could you please share me some code about "How to process the depth image in PYTHON in order to get the same depth value as Realsense Viewer?  "

Besides, i am appreciate if you could give me some suggestions about why I got different depth value using 0.001 scale and 0.0001 scale in python.

GuoCHance, 2023-09-13:
Another question, I found that changing Disparity shift make something strange in depth image .
[image]
Disparity Shift =0
[image]
Disparity shift =50
And you can see some strange pointcloud appear in front of the face.
It can be also observed in 2d depth image.

[image]
[image]

How to solve this problem?

GuoCHance, 2023-09-14:
My current situation:
1. Python scripts got different depth value from Realsense Viewer-> Don't know why. I closed the post-processing in viewer, but still can't got the same depth value in python. However, my work is aim to compute the relationship between marker and face (Both in python). So the depth value offset between python and viewer may not be seriours to me.
2.In my experient, i placed the head model in a fixed position, then placed the marker in different place. And I couldn't always get the correct relationship. I think this problem may be due to the inaccurate of depth value in different place. And I need to figure out how to solve this problem now.
3.Disparity shift lead to ghost noise.

I am trying to calibrate my D415 cam. If there is any progress I will update this issue.

MartyG-RealSense, 2023-09-14:
Thanks very much for the detailed update.  Please do update further if there is progress.  Good luck!

GuoCHance, 2023-09-14:
After finishing On-Chip Calibration in Realsense Viewer. Cam's Health Check improved from 0.48 to 0. Depth values become more stable and less noise now.
However, the "ghost noise" still appeared while setting larger Disparity Shift.

Besides, still don't know why 2D Markers 6DoF error in some placements.( RGB Detect Corner and ROI->RGBD reconstruct pointcloud-> Fit plane -> Compute vertex and 6dof.)

The Depth image seems stable and correct, and the corner detection also seems good.

Confusing about if it is possible that depth become worse when target is placed farther away from camera's center line....

MartyG-RealSense, 2023-09-14:
It's good to hear that you achieved a significant improvement through calibration.

I note in your align_to line that you are using **align_to = rs.stream.depth**

In this instruction, the target of the alignment is defined.  So by using rs.stream.depth, your program is performing **color to depth** alignment instead of depth to color alignment.  Depth to color alignment is **align_to = rs.stream.color**, whilst color to depth alignment is very rarely used.   So it would be worth changing this line to see how it affects your accuracy.

GuoCHance, 2023-09-14:
> It's good to hear that you achieved a significant improvement through calibration.
>
> I note in your align_to line that you are using **align_to = rs.stream.depth**
>
> In this instruction, the target of [...]

Thanks for suggestions!
I already replace the rs.align to perform Depth to color alignment.
However, all pointclouds are reconstructed in the same coordinate(no matter color coord. or depth coord.). I am still trying to figure out why different Marker placement lead to different 6dof relationship result.

Thanks again for your patient help!  :）

MartyG-RealSense, 2023-09-14:
If you are placing marker tags on an object then the invisible IR dot pattern cast by the projector may make tags more difficult for the camera to read.  So you could try disabling the emitter using the Python code at https://github.com/IntelRealSense/librealsense/issues/10065#issuecomment-994776524 to see whether the dot pattern might be influencing the readability of the tags.

GuoCHance, 2023-09-18:
Hi,MartyG!
Have tried your suggestion about disabling emitter, but it was not good for my situation. Maybe due to the brightness of environment was not so good, disabling emitter made my depth image worse( less valid depth value.)

I have a question now.
I placed the Marker and the Head Model stationary in the scene. And I export mesh 1. from Realsense Viewer.
Then I move the camera some distance in order to let the Marker not direct to camera. And export mesh 2. again.
[image]
mesh1
[image]
mesh2.(Move camera some distance and the Marker now is not directly in front of the camera)

Then I mesure the distance from Marker to Head model.(Have checked the points-pick and both mesh are in good work distance of camera about 0.5m)

And the result :
[image]
you can see that some point-sets have about 2mm difference in distance.

I tried many times and I thinks the offset is not due to the Depth value fluctuation.

So I am curious about :
Is it normal that the camera get different distance measurement when placing in different perspective?
In my work, I need to reconstruct pointclouds of Marker and Head Model, then compute their relative 6dof pose.
When I placed the Camera and Head Model in a fixed position and moved Marker in different position(for example, Left front, right front and right front). I got different relative pose results like I mentiond above(the point-sets distance).

I have already tried to improve the depth image quality.However it still not solve my problem.
Now I think that my situation maybe due to the limitation of the camera's accuracy ?
Or is there still something wrong of my operation?
Need suggestions to decide if I should keep trying to figure out the reason of different relative 6dof pose computation.
Or maybe I should improve my method.

MartyG-RealSense, 2023-09-18:
There is certainly the possibility that coordinates that are off to the side of the center of the camera's view may have slightly less accuracy than the coordinates at the center.  

Accuracy should increase as the **baseline** (the distance between the left and right sensors on the front of the camera) is increased.  So the D415 has less depth noise than the D435 models because the D415's left and right sensors are located further apart.  The D455 has even better accuracy because there is a larger horizontal distance than D415 between its left and right sensors, giving the camera a wide field of view that enables it to see more of a scene.

The baseline of a particular RealSense camera model is fixed and so there is no way to take advantage of a wider baseline without changing to a camera model with a wide baseline, unfortunately.

*******

In regard to what you can try with software changes instead of a hardware change, I noticed in your script that the line **frames = align.process(frames)** is placed before the post-processing filters are applied with the **process** lines.  Intel recommend that alignment is processed **after** the post-processing filters are applied.

MartyG-RealSense, 2023-09-18:
Under some circumstances, the inaccuracy can become much larger than 2 mm the further that the coordinates are from the center.  However, this usually occurs only when using alignment and also using the incorrect type of intrinsics.  

For example, if depth to color alignment is used then **color** instrinsics should be used, and if color to depth alignment is used then **depth** intrinsics should be used.   This problem with large measurement inaccuracies does not occur if alignment is not used.

GuoCHance, 2023-09-18:
Fine.
I experient the meshes which are exported from Realsense Viewer and I don't think the software use the incorrect intrinsics.
Seems I still need to figure out what happen or just stop because of the inaccuracy when coordinates are off to the side of the center.

Besides, My python code use rs.align and I think I didn't use the wrong intrinsic.
Thanks.

MartyG-RealSense, 2023-09-18:
Would it be possible to make the marker board smaller and attach it to the forehead of the dummy head?

GuoCHance, 2023-09-18:
Sure. 
Actually when marker board is placed in front of the dummy head's nose. Their relative 6dof pose could be calculated correct. But I think this has limitation in real situations, so I try to test if it is work when marker is placed a distance from the  dummy head. Unfortunately can't reach my precision requirement now.

MartyG-RealSense, 2023-09-18:
If perspective is causing the image to appear to the camera to be moving further sidewards away from the head as the camera is moved further away, then it might help if the image is made smaller and moved closer to the neck in front of the head.  Then when the camera is moved away, the image may still move sidewards as the perspective changes but not enough to cause a 2 mm inaccuracy.

GuoCHance, 2023-09-19:
Thanks a lot, MartyG!
I have no more question now and I will close this issue.
Appreciate for your quick and patient suggestion

MartyG-RealSense, 2023-09-19:
You are very welcome.  I'm pleased that I could help  :)

closed 2023-09-19 by MartyG-RealSense