realsenseai/librealsense #12184: GMSL D457 Cannot set exposure time

nathanieltagg, 2023-09-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R457 (GMSL connectors) | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Ubuntu 20.04 (Jetpack 5.0.2)  | 
| Kernel Version (Linux Only)    |  5.15.104                                      | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            | R/254   and development                 | 
| Language                            |  C++                          | 
| Segment			|  Robot                           | 

### Issue Description
The camera allows manual exposure to be set (in the realsense viewer and in software) appears to allow valid exposure times only in the range [0,0], so any manual setting changes to the minimum exposure time and almost-blacked-out images in our lab setting. This happens on both 5.13 and 5.15 firmware, and in both release and development versions of the library.

Our application requires minimum motion blur, so manual control of the exposure is important.  Is this a known issue or limitation with the MIPI interface?  If not, any suggestions for remediation?

MartyG-RealSense, 2023-09-13:
Hi @nathanieltagg  We have not had a previous report of the described issue, though there are not many MIPI users of the D457 because owners typically use the camera in its USB mode.  I do not have a MIPI setup to test my D457 with in order to replicate your findings, unfortunately.

Motion blur should not be a problem on D457 though because both the RGB and depth sensors are equipped with fast 'global' shutters that can process high-speed motion of the camera or observation of fast-moving objects.  Increasing FPS speed above 30 FPS (such as 60 or 90) can also help to reduce the likelihood of motion blur occurring.

nathanieltagg, 2023-09-20:
Hi @MartyG-RealSense,
That statement is just... incorrect. We are working in an indoors setting, a typical warehouse environment with high flourescent lighting of maybe 10-20 lux.  In this environment, the automatic exposure time typically runs at something like 80 ms, which causes severe motion blur for even slow turns of our vehicle.  Global shutter means that there is no rolling-shutter distorition, but motion blur is still a real thing.

Increasing FPS doesn't actually help motion blur, and in fact the camera cannot run at 30 FPS because the exposure time is so long.  To combat this, we are currently manually adjusting exposure time and gain to find a 'happy middle' ground with poor color and reasonable blur.

In any case, the configuration, available in USB, is not available in GMSL for some reason.

MartyG-RealSense, 2023-09-20:
Fluorescent lights such as ceiling strip lights can cause interference to RealSense images as they contain gas that flickers at rates that are difficult to see with the human eye.  

Configuring the **Power Line Frequency** RGB option can help to deal with this.  It can be set to 50 or 60.  Lights in European regions typically use 50 Hz power line frequency, whilst North America uses 60 Hz.

[image]

MartyG-RealSense, 2023-09-28:
Hi @nathanieltagg  Do you require further assistance with this case, please?  Thanks!

nathanieltagg, 2023-09-28:
Yes, the issue is still outstanding. Over GMSL many controls are disabled and exposure time control does not work.

MartyG-RealSense, 2023-09-29:
Which model of Jetson and which deserializer board are you using, please?  

Intel's MIPI driver was designed for use specifically with Jetson AGX Xavier and Intel's own brand of deserializer board.  If you use a computing device and deserializer other than these ones then the MIPI driver has to be converted (ported) for use with your particular equipment.  Intel are unfortunately unable to assist with this porting process.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

nathanieltagg, 2023-09-29:
This is a ConnectTec Anvil Orin AGX.  

However, it looks to me like the RS code is at least partly involved, since it reports that the valid exposure range is [0,0].   That implies that at the very least a communication error isn’t being reported back to the user.  

Other problems with this system include the multi cam problem I reported in a separate issue.

MartyG-RealSense, 2023-09-29:
The Connect Tech system does not require a separate deserializer.  

https://connecttech.com/connect-tech-offers-built-in-support-on-the-newly-announced-intel-gmsl-realsense-depth-camera-d457/

Connect Tech will be the appropriate people to contact for technical support in this particular case because of its dedicated built-in support for D457, as Intel are unable to provide technical support for RealSense-compatible hardware and software products provided by other companies.

There is a discussion about using Connect Tech with D457 in MIPI mode at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1486939651

MartyG-RealSense, 2023-10-05:
Hi @nathanieltagg  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!

dmipx, 2023-10-09:
Hi. Did you disabled "Auto Exposure" option before setting manual exposure?

closed 2023-10-15 by MartyG-RealSense