realsenseai/librealsense #12185: D405 Setting Exposure in OpenCV

DaveBGld, 2023-09-12:
Using `cap.set(cv2.CAP_PROP_EXPOSURE, exposureValue)`  no matter what exposure value (i.e. -4, 100000) does nothing.

I have had to resort to this code:

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
pipeline.start(config)
depth_sensor = pipeline.get_active_profile().get_device().first_depth_sensor()
exposureValue = 100000
depth_sensor.set_option(rs.option.exposure, exposureValue)
pipeline.stop()

camera_index = 4
cap = cv2.VideoCapture(camera_index)
```
But this code **slows the fps to 6 instead of 30....**

Specific question: What is the preferred way to set the exposure value in OpenCV Python?

Thanks!

DBG

MartyG-RealSense, 2023-09-13:
Hi @DaveBGld  Your FPS is lagging because of the '100000' exposure value that you have set.  When setting a manual exposure value, certain value ranges can cause FPS to slow, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

The default exposure value for the D405 is '33000'.

The D405 does not have a separate RGB sensor and so shares its exposure setting with depth.  So whatever the depth exposure is manually set to will affect RGB color exposure too.  I confirmed in my own tests that FPS is 30 at exposure 33000 and drops to 6 at 100000 exposure.

OpenCV has an exposure command called **CAP_PROP_EXPOSURE** for altering image brightness.

https://www.kurokesu.com/main/2020/05/22/uvc-camera-exposure-timing-in-opencv/

DaveBGld, 2023-09-13:
@MartyG-RealSense Thanks for the reply

I tried this:

```
cap = cv2.VideoCapture(camera_index)
cap.set(cv2.CAP_PROP_AUTO_EXPOSURE, 0.25) # manual mode
cap.set(cv2.CAP_PROP_EXPOSURE, -10) # This should return a black frame

```
And also 

`cap.set(cv2.CAP_PROP_AUTO_EXPOSURE, 0.75) # auto mode
`

before setting manual mode

And also using values 3 and 1 for auto and manual exposure.

Nothing works.

The OpenCV forums point out that these values are back-end (Intel libreaalsense) dependent. People report different values for different cameras...

DaveBGld, 2023-09-14:
@MartyG-RealSense  thanks for that.

Still the code you suggested 

```
cap = cv2.VideoCapture(camera_index)
cap.set(cv2.CAP_PROP_AUTO_EXPOSURE, 0.25) # manual mode
cap.set(cv2.CAP_PROP_EXPOSURE, -10) # This should return a black frame
```
doesn't appear to have any effect on the camera feed.

Any other ideas?

Thanks,

DBG

MartyG-RealSense, 2023-09-14:
What if you set VideoCapture as '0' instead of camera_index?

`cap = cv2.VideoCapture(0)`

DaveBGld, 2023-09-19:
@MartyG-RealSense thanks, but I have two cameras connected

MartyG-RealSense, 2023-09-19:
I do not have any other suggestions to offer, unfortunately.  I do apologize.

DaveBGld, 2023-09-19:
So the D405 camera is not compatible with OpenCV...

MartyG-RealSense, 2023-09-19:
RealSense cameras are fully compatible with OpenCV.  

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv

https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

Typically, auto-exposure is used with RealSense OpenCV applications rather than manual exposure though.

closed 2023-09-20 by MartyG-RealSense