realsenseai/librealsense #12188: Missing wheels for arm architecture for python 3.10 and 3.11

MikiGrit, 2023-09-13:
### Issue Description

On the pypi site (https://pypi.org/project/pyrealsense2/#files), the pyrealsense2 does not include wheels for python 3.10 and 3.11 for arm architecture (`aarch64`). Do you plan to publish those? It would make my life much easier.

Thanks.

MartyG-RealSense, 2023-09-13:
Hi @MikiGrit  I will enquire to my Intel RealSense colleagues about aarch64 wheel files for Python 10 and 11.

MartyG-RealSense, 2023-09-29:
Hi again @MikiGrit  My colleagues responded that older versions of Python were chosen to create wheels for is because those are the default Python versions that come with "JP4 and JP5" (this is likely a reference to Nvidia Jetson's JetPack 4 and JetPack 5).

They are open to the possibility of adding wheel support for newer versions in future once they see how it would affect development resources.

MikiGrit, 2023-10-03:
Ok, thanks for the response. We actually have very similar use-case. We are using JetPack 5. But due to compatibility with other python libraries, we are forced to use more recent python. It would be great if they could find the time to support also newer python versions.

MartyG-RealSense, 2023-10-11:
Hi @MikiGrit  As wheel files for Python 3.10 and 3.11 are not being developed at this time (although it is a possibility for the future), can this case be closed?  Thanks!

MikiGrit, 2023-10-11:
@MartyG-RealSense Well, sure. I don't have anything other in mind. Thanks for your effort anyway.

MartyG-RealSense, 2023-10-11:
@MikiGrit  You are very welcome.  Thanks very much for your understanding.

closed 2023-10-11 by MartyG-RealSense