realsenseai/librealsense #12189: Align already saved color images to depth images

songdedaoyi, 2023-09-13:
Hi, guys.  I already have captured color and depth images (as .tif) but I noticed that they are not aligned. However, I need to align color images to depth images. I have intrinsics and extrinsics data but I don't know how to properly use them - all I've seen so far is processing images from the stream, not from the file.
Can please someone navigate me or link the similar problem?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10) / Linux (Ubuntu 14/16/17) | 
| Platform                                 | PC/NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                          | 
| Segment			|  others }                          |

MartyG-RealSense, 2023-09-13:
Hi @songdedaoyi  Except for the **.raw** image file format, saving depth information to image file will cause most of the depth information to be lost and so it is not recommendable to do it.

The only available reference that discusses using Python to align TIFF image files is at https://github.com/IntelRealSense/librealsense/issues/1274

songdedaoyi, 2023-09-14:
> Hi @songdedaoyi Except for the **.raw** image file format, saving depth information to image file will cause most of the depth information to be lost and so it is not recommendable to do it.
> 
> The only available reference that discusses using Python to align TIFF image files is at #1274
Thanks a lot！However, I am still confused that whether it is possible to align already saved color images to depth images since the issues I had read are about how to align depth images to color images.

closed 2023-09-27 by MartyG-RealSense