realsenseai/librealsense #12194: (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument

zhongwei8, 2023-09-14:
ERROR: (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument 
---------------------------------------------------------------------------------------------------------------------------------------------
14/09 11:12:30,120 INFO [281473548606064] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 14/09 11:12:30,120 INFO [281473548606064] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 14/09 11:12:30,120 INFO [281473548606064] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 14/09 11:12:30,151 INFO [281473548606064] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 14/09 11:12:30,299 INFO [281473548606064] (context.cpp:336) Found 4 RealSense devices (mask 0xff)
 14/09 11:12:30,353 INFO [281473548606064] (rs.cpp:2697) Framebuffer size changed to 1344 x 586
 14/09 11:12:30,354 INFO [281473548606064] (rs.cpp:2697) Window size changed to 1344 x 586
 14/09 11:12:30,354 INFO [281473548606064] (rs.cpp:2697) Scale Factor is now 1
 14/09 11:12:30,359 ERROR [281473422729584] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 14/09 11:12:30,361 WARNING [281473422729584] (rs.cpp:312) null pointer passed for argument "device"
 14/09 11:12:30,361 WARNING [281473422729584] (rs.cpp:2700) Couldn't refresh devices - xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 14/09 11:12:30,833 INFO [281473548606064] (context.cpp:336) Found 4 RealSense devices (mask 0xfe)

MartyG-RealSense, 2023-09-14:
Hi @zhongwei8  A **xioctl(VIDIOC_S_EXT_CTRLS)** indicates that there may be a conflict with the Linux kernel.  Which kernel version are you using, please?

zhongwei8, 2023-09-18:
20.04

| |
***@***.***
|
|
***@***.***
|

---- Replied Message ----
| From | ***@***.***> |
| Date | 09/14/2023 18:32 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [IntelRealSense/librealsense] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument (Issue #12194) |

Hi @zhongwei8 A xioctl(VIDIOC_S_EXT_CTRLS) indicates that there may be a conflict with the Linux kernel. Which kernel version are you using, please?

MartyG-RealSense, 2023-09-18:
20.04 is the Ubuntu version.  The kernel version would be numbered something like 5.15 or 5.19.

MartyG-RealSense, 2023-10-01:
Case closed due to no further comments received.

closed 2023-10-01 by MartyG-RealSense

sriramreddyM, 2023-11-02:
@MartyG-RealSense I get same error on Nvidia Jetson Nano. Using Jetpack 4.6, Kernel version: 4.9.253-tegra.
What could be the issue?

MartyG-RealSense, 2023-11-02:
Hi @sriramreddyM  Which Ubuntu version are you using, please?  JetPack 4.6 is designed for use with Ubuntu **18.04**, whilst Ubuntu 20.04 should be used with JetPack 5.

sriramreddyM, 2023-11-02:
Hi,

Yes, I'm Ubuntu 18.04 version.

MartyG-RealSense, 2023-11-02:
@sriramreddyM  Did you install the librealsense SDK from source code, please?  If you did then which installation method did you use?

sriramreddyM, 2023-11-02:
I tried both ways from https://github.com/JetsonHacksNano/installLibrealsense

MartyG-RealSense, 2023-11-02:
The notes for the source build method **installLibrealsense** state that "the build uses libuvc. You will not have to rebuild the kernel or modules in order to use this build".  

A libuvc build of librealsense bypasses the Linux kernel and does not require a kernel patch script to be applied to the kernel  It seems unusual therefore that you are experiencing the **xioctl(VIDIOC_S_EXT_CTRLS)** error, as that indicates a conflict with the kernel that should not be occurring with a kernel-bypassed SDK build.

Ideally though there should not be both package and source code builds on the same computer (an installLibrealsense package build and a buildLibrealsense source code build) as there may be a conflict between the two different types of installation.

Please try removing all RealSense related packages on the computer with the Ubuntu terminal instruction below so that only the source code build of librealsense is left remaining.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Does removing RealSense packages make a difference to whether the **xioctl(VIDIOC_S_EXT_CTRLS)** error appears?

WeiShengAng, 2023-12-08:
@MartyG-RealSense I met the same problem since last week, but thanks to your latest asnwer, it does solve my problem! After removing all Realsense related pkg and install realsense2-camera pkg again, it works perfectly !!!

alxhoff, 2025-04-09:
Hi I am getting this xioctl(VIDIOC_S_EXT_CTRLS) error on my Jetson. I see on the releases notes that a 5.10 kernel is not listed as being supported. Which is strange since the kernel side GMSL driver supports jetpack 5.1.2 which is a 5.10 kernel. 

I am using v2.56.3 of librealsense.

Assuming this is an oversight I have manually patched my kernel with 

01-realsense-camera-formats-L4T-5.0.2.
02-realsense-metadata-L4T-5.0.2.patch
05-realsense-powerlinefrequency-control-fix.patch

as this seems to be the patches applied by patch-realsense-ubuntu-L4T.sh

@MartyG-RealSense can you please let me know if there is anything else I should be doing to build a 5.10.X kernel to remove this error.

MartyG-RealSense, 2025-04-09:
Hi @alxhoff  Jetson boards have a different set of installation instructions to the ones for PC.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md

With Jetson, the **JetPack** version is the most important factor in a librealsense installation instead of the kernel version.  So as long as you have a compatible JetPack installed, it does not matter if the kernel version is 5.10.

The 2.53.1 version of librealsense has support for JetPack 6.  JetPack 5.1.2 is also compatible, as you mention.

If librealsense is built from source code with CMake with the **libuvc backend** or **RSUSB backend** installation methods then it is not necessary to apply the L4T patch script, as librealsense bypasses the kernel when built using either of these two methods.

alxhoff, 2025-04-09:
Thanks for the quick reply @MartyG-RealSense , I have followed the page you linked but I am using the realsense MIPI driver as I am using GMSL cameras and not USB. I have no been able to find a solution to the similar VIDIOC_S_EXT_CTRLS ioctl issue for non-usb cameras. Do you have any experience with this?

MartyG-RealSense, 2025-04-09:
GMSL is a specialist subject, so I recommend posting on the dedicated support forum for the MIPI driver to receive advice from my Intel RealSense colleagues of who have GMSL expertise.

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues

On that forum, a RealSense GMSL user at https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/255 also experienced a similar (though slightly different) VIDIOC error -theirs was G-EXT instead of S-EXT.

alxhoff, 2025-04-09:
Thanks I will post there. Appreciate your quick responses