realsenseai/librealsense #12198: Read ideal max depth distance

andrewwongagility, 2023-09-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.14 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC |
| SDK Version                            |  2.5.0                          | 
| Language                            |  C++                           | 
| Segment			|  Robot                           | 

### Issue Description
I'd like to read (not modify) the max depth range of a RealSense D455 through the librealsense API. For example, the D455 ideal max range is [6 meters](https://www.intelrealsense.com/depth-camera-d455/). Is this possible? 

So far, I've tried:
1.`sensor.as<rs2::max_usable_range_sensor>().get_max_usable_depth_range().`   
I get a runtime error saying it failed to convert `sensor` to a `max_usable_range_sensor`. I think the D455 may not be able to be converted to a `max_usable_range_sensor`. The [release notes for librealsense 2.4.0](https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2400)  imply this function is specific to the L515.

2. `sensors.get_option(RS2_OPTION_MAX_DISTANCE)`. I get a runtime error saying this option is not supported.

3. `const auto table = sensor.as<rs400::advanced_mode>().advanced.get_depth_table().` 
`table.depthClampMin` and `table.depthClampMax` are 0 and 65536, respectively, which seem to be uninitialized values.

 Any help is greatly appreciated!

MartyG-RealSense, 2023-09-14:
Hi @andrewwongagility  Yes, **get_max_usable_depth_range** is an L515 feature.  If you are not modifying settings though then the ranges of the cameras are relatively fixed.

Most cameras in the 400 Series range have a depth sensing range of 10 meters.

Although D455 has an ideal depth range of up to 6 meters, it can depth sense up to 20 meters.  The link below demonstrates this.

https://support.intelrealsense.com/hc/en-us/community/posts/17214523195155-Camera-for-power-line-inspection

D405 has an ideal depth range of 7 cm to 50 cm as it is designed for close-range high accuracy, high quality sensing but can render depth up to a few meters.

L515 has a range of 9 meters, though the best results will be with distances below 4 meters.

In regard to setting a custom maximum depth distance, you can do this easily by configuring the post-processing filter **Threshold Filter**.

andrewwongagility, 2023-09-14:
Thank you @MartyG-RealSense ! That's very helpful info. 

We're using several different Realsense D400 models (e.g. D430, D450), and it would be nice to get the max ideal depth range from the librealsense API without knowing the D400 model type. Do you know of any API calls that can do this?

MartyG-RealSense, 2023-09-14:
Defining a Threshold Filter and setting its maximum depth to 10 meters will likely work best for you.  Then D450 will be forced to limit itself to rendering the same maximum distance as D430.

The link below has a C++ librealsense script that makes use of the Threshold Filter.

https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds

andrewwongagility, 2023-09-14:
Thank you @MartyG-RealSense! I'll take a look. After speaking with my team more, I think we may want to hard code max depth based on the specs Intel provides for each RealSense camera. I really appreciate your help!

closed 2023-09-14 by andrewwongagility

MartyG-RealSense, 2023-09-14:
You are very welcome!

You could consider reading the name of the camera with the instruction **RS2_CAMERA_INFO_NAME** and setting different configurations based on a particular camera name, like in the C++ script at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468

```
for (auto dev : ctx.query_devices()) // For each device do :
{
if (strcmp(dev.get_info(RS2_CAMERA_INFO_NAME), "Intel RealSense D415") == 0)  // Check for compatibility
{
```