realsenseai/librealsense #12199: Pose-processing (temporal filter) causes issues pyrealsense2

jashshah999, 2023-09-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 } | 
| Firmware Version                   | (2.51.1) | 
| Operating System & Version |   {Win10} | 
| Kernel Version (Linux Only)    |  NA                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.51.1 }                          | 
| Language                            |  {Python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
This is my script

`import cv2 
import numpy as np
import pyrealsense2 as rs

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30) 

# Enable temporal filter
filter = rs.temporal_filter() 
filter.set_option(rs.option.filter_smooth_alpha, 0.66)
filter.set_option(rs.option.filter_smooth_delta, 46)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30, filter)

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()

        # Get depth frame
        depth_frame = frames.get_depth_frame() 

        # Filter already applied by RealSenseSDK
        
        # Convert depth to NumPy array
        depth_image = np.asanyarray(depth_frame.get_data())
        
        # Apply colormap
        depth_colormap = cv2.applyColorMap(
            cv2.convertScaleAbs(depth_image, alpha=0.03), 
            cv2.COLORMAP_JET)

        # Show depth image
        cv2.imshow('RealSense', depth_colormap)
        cv2.waitKey(1)

finally:
   
    # Stop streaming
    pipeline.stop()`
    

When I do not enable the temporal filter, everything works fine and I can see the stream. However, when I enable the temporal filter, it throws this error

`    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30, filter)                                          TypeError: enable_stream(): incompatible function arguments. The following argument types are supported:                    1. (self: pyrealsense2.pyrealsense2.config, stream_type: pyrealsense2.pyrealsense2.stream, stream_index: int, width: int, height: int, format: pyrealsense2.pyrealsense2.format = <format.any: 0>, framerate: int = 0) -> None                  2. (self: pyrealsense2.pyrealsense2.config, stream_type: pyrealsense2.pyrealsense2.stream, stream_index: int = -1) -> None                                                                                                                      3. (self: pyrealsense2.pyrealsense2.config, stream_type: pyrealsense2.pyrealsense2.stream, format: pyrealsense2.pyrealsense2.format, framerate: int = 0) -> None                                                                                4. (self: pyrealsense2.pyrealsense2.config, stream_type: pyrealsense2.pyrealsense2.stream, width: int, height: int, format: pyrealsense2.pyrealsense2.format = <format.any: 0>, framerate: int = 0) -> None                                     5. (self: pyrealsense2.pyrealsense2.config, stream_type: pyrealsense2.pyrealsense2.stream, stream_index: int, format: pyrealsense2.pyrealsense2.format, framerate: int = 0) -> None                                                                                                                                                                                 Invoked with: <pyrealsense2.pyrealsense2.config object at 0x000002B7DE4F0D70>, <stream.depth: 1>, 640, 480, <format.z16: 1>, 30, <pyrealsense2.pyrealsense2.temporal_filter object at 0x000002B7EF56A7B0> `

Please guide me as to what needs to be done. I am using Usb3.2 (confirmed via intel realsense viewer). Thank you.

MartyG-RealSense, 2023-09-21:
Hi @jashshah999  Do you require further assistance with this case, please?  Thanks!

jashshah999, 2023-09-21:
Hi, for some reason, pressing 'f' is not working. If I use a different filter like rs.hole_filling_filter, it works but when I run the following code, pressing 'f' makes no difference (the temporal filter is not working for some reason)

```
import cv2
import numpy as np
import pyrealsense2 as rs

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# Start streaming
pipeline.start(config)

apply_filter = True

try:
    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()

        # Get depth frame
        depth_frame = frames.get_depth_frame()

        if apply_filter:
            # filter = rs.hole_filling_filter(1)
            # filter = rs.hole_filling_filter(2)
            # Adjust filter parameters when 'f' is pressed
            filter = rs.temporal_filter()
            filter_smooth_alpha = 0.66  # Change alpha value
            filter_smooth_delta = 46   # Change delta value
            filter.set_option(rs.option.filter_smooth_alpha, filter_smooth_alpha)
            filter.set_option(rs.option.filter_smooth_delta, filter_smooth_delta)
            # filter.set_option(rs.option.persistency_filter,8)

            depth_frame = filter.process(depth_frame)

        # Convert depth to NumPy array
        depth_image = np.asanyarray(depth_frame.get_data())
        print(np.max(depth_image,axis=0))
        # Apply colormap
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # Show depth image
        cv2.imshow('RealSense', depth_colormap)

        key = cv2.waitKey(1)
        if key == ord('f'):
            apply_filter = not apply_filter
            print("Pressed filter change")

finally:
    # Stop streaming
    pipeline.stop()

```

MartyG-RealSense, 2023-09-22:
Does it make a difference if you move the filter settings outside of the `if apply_filter` condition so that the pressing of the 'f' key only determines whether or not the filter settings are applied to the image?

```
# filter = rs.hole_filling_filter(1)
# filter = rs.hole_filling_filter(2)
# Adjust filter parameters when 'f' is pressed
filter = rs.temporal_filter()
filter_smooth_alpha = 0.66  # Change alpha value
filter_smooth_delta = 46   # Change delta value
filter.set_option(rs.option.filter_smooth_alpha, filter_smooth_alpha)
filter.set_option(rs.option.filter_smooth_delta, filter_smooth_delta)
# filter.set_option(rs.option.persistency_filter,8)

if apply_filter:
depth_frame = filter.process(depth_frame)
```

closed 2023-10-05 by MartyG-RealSense