realsenseai/librealsense #12203: ModuleNotFoundError: No module named 'pyrealsense2'

tsaiboyu, 2023-09-18:
I installed librealsense with jetson nano nx and the installation process went smoothly, but when I ran the program, File "detect.py", line 10, in <module>
    import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'
I don't have this problem when using windows  pi
"echo $PYTHONPATH
/opt/ros/melodic/lib/python2.7/dist-packages:/usr/local/lib:/usr/local/lib/python3.6/pyrealsense2
"The search shows up, please help.

MartyG-RealSense, 2023-09-18:
Hi @tsaiboyu  A guide for building librealsense and pyrealsense2 from source code on Jetson together at the same time can be found at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

Using this guide's installation method recently helped another Jetson user at https://github.com/IntelRealSense/librealsense/issues/12112 to resolve the **No module named 'pyrealsense2'** problem.

tsaiboyu, 2023-09-21:
@MartyG-RealSense 
Thanks I solved it ＃12112, but I encountered a new problem,
"import pyrealsense2 as rs
ImportError: /usr/lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/boyu/catkin_ws/src/librealsense/build/wrappers/python/pyrealsense2.cpython-36m-aarch64-linux-gnu.so)"

tsaiboyu, 2023-09-21:
hi @MartyG-RealSense 
I have solved it , thank you very much^^

MartyG-RealSense, 2023-09-21:
That's great news that you achieved a solution, @tsaiboyu - thanks very much for the update!

closed 2023-09-28 by MartyG-RealSense