realsenseai/librealsense #12205: D455 - Invalid SKU

arpaterson, 2023-09-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.13.50, 5.12.7.150| 
| Operating System & Version |  Win11 and Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |  doesn't matter                      | 
| Language                            | doesn't matter                        | 
| Segment			|  Robot/others }                          | 

### Issue Description

This is basically an additional case of issue [11621](https://github.com/IntelRealSense/librealsense/issues/11621).

The camera reports a "Device Descriptor Request Failed" in Windows.
In Ubuntu `dmesg` reports an "Invalid SKU".

No intel tools work.
Realsense Viewer will not interact with the camera at all, or gives an error under linux.
Realsense firmware updater also will not recover the camera, regardless of whether the D450 board is connected to the D4 board. 

With the D450 board disconnected, it shows up as a D405i tester, and rs-viewer and rs-fw-updater will still not update the board **(perhaps this can be remedied with an update to those tools, to allow users to recover cameras without extra hardware/return/trashing). The device descriptors for invalid SKU/tester need to be added and handled appropriately**

Similar to issue 11621, I was able to recover this camera buy connecting an older D450 board marked K83121-100. I was not able to confirm whether the original D450 is K83121-101 which is the issue as it is missing the sticker, but I suspect it is. **The "-101" board does not work with older firmwares. This is the issue.**

I also bought an extra D4 alongside the D450 and HSF50 ribbon cable:

One D4 is marked PC0603902 and had firmware 5.12.13.50. With the older D450 "-100" attached I was able to update to 5.13.0.50 (for realsense-ros 2.3.2).

The other is marked PC0603901, this was also not able to be updated until the older D450 "-100" board was connected. It had 5.12.7.150, and was also updated to 5.13.0.50. 

So it appears the issue is with firmwares older than 5.13.0.50 at least. 
Cameras should not be downgraded too far if they have the "-101" module.

The firmware update was done under windows with the original USB3 cable, and was an update `Intel.RealSense.FW.Update.exe -f <bin FILE>`, not a recovery `-r`

After the update, both boards are recognized under windows..

Needed items: 
- D450 V3 (K83121-100 not -101!!) 
- and a HSF50 ribbon cable to avoid dismantling the D450 further as there is adhesive involved. The one in the camera is too short, stiff and awkward, and you risk shorting the board by not using an extra ribbon cable and removing the D4 board from the camera. Just get it.

MartyG-RealSense, 2023-09-18:
Hi @arpaterson  Thanks so much for your detailed feedback that verifies the solution of updating the firmware using an older board.  

Intel are working on adding support for 'D405i Tester' to the librealsense SDK to allow firmware update in this mode, so this invalid SKU issue should hopefully be fixable in future without having to obtain an older board.

arpaterson, 2023-09-18:
Thanks @MartyG-RealSense. Thats both obvious and fantastic. 

I tested under Ubuntu and windows and the camera works with realsense viewer now.

There were some errors on startup 
`motion module failure`
 and 
`Invalid Value in rs2_get_option(options:0DB933A8, option:Emitter On Off):
hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)`.

However, I am using a newer version of realsense viewer, with an older firmware. perhaps that is the reason?

Otherwise, it seems ok. The IMU data is actually coming thru, so I don't know what motion module failure means.

arpaterson, 2023-09-18:
I would one more thing regarding disassembling the camera.
Without the case, the ASIC does not have sufficient cooling, so test very briefly if at all without the D455 housing.

MartyG-RealSense, 2023-09-18:
Motion Module Failure typically means that there was a momentary glitch where the IMU was not responsive.  The IMU usually functions if it is enabled again after the error has occurred, so the issue is more of a temporary annoyance than a serious problem.  

Whilst it used to be possible to use recent firmwares with older SDK versions, that is no longer the case and there is usually one specific firmware version that is recommended to be used with a particular SDK version, otherwise errors occur.

2.54.1: firmware 5.15.0.2
2.53.1: firmware 5.14.0.0
2.50.0 and 2.51.1: firmware 5.13.0.50

In regard to disassembly, RealSense cameras have a block with radiator fins called a **heat sink** mounted on the rear of the depth camera module.  A heat sink draws heat away from electronics.  So the removal of the heat sink would certainly have an impact on the ability of the camera hardware to be kept cool during operation.

[image]

arpaterson, 2023-09-29:
> Hi @arpaterson Do you require further assistance with this case, please? Thanks!

No, thank you. The camera was successfully recovered. I will not use any firmware below 5.13

MartyG-RealSense, 2023-09-29:
Thanks very much for the update.  As you have had a successful outcome, I will close this case.  Thanks again!

closed 2023-09-29 by MartyG-RealSense