realsenseai/librealsense #12206: No update from the depth image

WhoIsChrisss, 2023-09-18:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|243522072466|
|**Firmware Version**|5.15.0.2|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|238623060583|
|**Firmware Update Id**|238623060583|

Hello,

I'm using the "Intel D435i Depth camera" together with windows and python. I have the problem that the depth image is not updated and only the first frame is displayed. It is also very strange that the depth image is displayed smoothly in the real sense viewer. With the RGB-Image everything is fine. I tried the following example program:

`## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

###############################################
##      Open CV and Numpy integration        ##
###############################################

import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        depth_colormap_dim = depth_colormap.shape
        color_colormap_dim = color_image.shape

        # If depth and color resolutions are different, resize color image to match depth image for display
        if depth_colormap_dim != color_colormap_dim:
            resized_color_image = cv2.resize(color_image, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]), interpolation=cv2.INTER_AREA)
            images = np.hstack((resized_color_image, depth_colormap))
        else:
            images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()`

I also tried different versions of the librealsense2-python wrapper and different firmware version of the camera, but its all the same. 

Please help me...

MartyG-RealSense, 2023-09-18:
Hi @WhoIsChrisss   Your script is identical to the example program **opencv_viewer_example.py** and so it is not clear why the depth stream would not be working correctly.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

When you launch the script, are there any other RealSense programs such as RealSense Viewer that are already running and streaming camera data?  The Python script should be the only RealSense program that is running.  This is because the camer's sensors can only be accessed by one program at a time and lock out access by any other program that is subsequently launched after the first program that is using the sensor.

Have you tried the example **align_depth2color.py** to see whether depth update also freezes in that program, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

WhoIsChrisss, 2023-09-19:
Problem solved after using the USB 3.0 Port and not the 2.0...
Feel a little bit stupid. Thanks for help. :-)

MartyG-RealSense, 2023-09-19:
It's no trouble at all.  It's easy to happen.  :)  Thanks very much for the update!