realsenseai/librealsense #12208: Framos network enabled D455 fails to serve streams

gyillikci, 2023-09-19:
| | |
|---|---|
|**librealsense**|2.50.12 RELEASE|
|**OS**|Windows|

Framos d455 network enabled device throws out this error, and the app crushes
[image]
Any Idea why would this happen

Best

MartyG-RealSense, 2023-09-19:
Hi @gyillikci  Intel are unable to provide technical support for the FRAMOS range of 'e' industrial RealSense cameras unfortunately as they are not an Intel product.  Please visit the FRAMOS support page at the link below and click the 'Contact Us' button.

https://www.framos.com/en/resources/support

gyillikci, 2023-09-19:
Thanks Marty, 

on their interface, there is a tab which directs to here for issues. that's why I wrote to here. Maybe you guys resolve that with Framos.

closed 2023-09-19 by gyillikci