realsenseai/librealsense #12211: The recorded .bag file is 30fps, but the read .bag file is 10fps

puyiwen, 2023-09-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 5.14.0 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |python}                          | 
| Segment			| others }                          | 

### Issue Description
I record the .bag set 30fps, but I read the .bag file, I use `get_timestamp() ` find the frames interval 100ms, it means 10fps, not the recorded file I set. I am sure the .bag file has already set 30fps. I dont know why, can you help me? Thank you very much!!

Here my code
`#coding=utf-8
import pyrealsense2 as rs
import numpy as np
import cv2
import os
import PIL
from PIL import Image
import matplotlib.pyplot as plt

ALIGN_WAY=1
 
def Align_version(frames,align,show_pic=0,save_pic=0):
    aligned_frames = align.process(frames)
    depth_frame_aligned = aligned_frames .get_depth_frame()
    color_frame_aligned = aligned_frames .get_color_frame()
    return color_frame_aligned,depth_frame_aligned
 
    
def save_image_results(prediction,save_depth_path): 
    print(prediction.shape)
    vmin = prediction.min()
    vmax = prediction.max()
    save_to_dir = save_depth_path
    fig = plt.figure(figsize=(6.40, 4.80),dpi=100)
    ax = plt.Axes(fig, [0., 0., 1., 1.])
    ax.set_axis_off()
    fig.add_axes(ax)
    pred_map = ax.imshow(prediction, vmin=vmin, vmax=vmax, cmap='viridis')
    fig.colorbar(pred_map, ax=ax, shrink=0.8)
    fig.savefig(save_to_dir)
    plt.clf()
 
if __name__ == "__main__":
    # Configure depth and color streams
    pipeline = rs.pipeline()
    config = rs.config()
 
    config.enable_device_from_file("/home/puyiwen/Downloads/20230919_162605.bag")
    if ALIGN_WAY:
        way=rs.stream.color
    else:
        way=rs.stream.depth
    align = rs.align(way)
    profile =pipeline.start(config)
 
 
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_scale = depth_sensor.get_depth_scale()
    print("scale:", depth_scale)
    save_path = '/home/puyiwen/Downloads/shumeipai/test'
    if not os.path.exists(save_path):
        os.mkdir(save_path)
    idx = 1
    print("Frames Captured")
    colorizer = rs.colorizer()
    while True:
        frames = pipeline.wait_for_frames()
        color_image_aligned_org,depth_image_aligned_org=Align_version(frames,align)
        depth_frame_time = depth_image_aligned_org.get_timestamp()
        depth_frame_time = depth_frame_time / 1000.0
        color_frame_time = color_image_aligned_org.get_timestamp()
        color_frame_time = color_frame_time / 1000.0
        print('depth_frame_time',depth_frame_time)
        print('color_frame_time',color_frame_time)
        color_image_aligned = np.asanyarray(color_image_aligned_org.get_data())
        color_image_aligned = cv2.cvtColor(color_image_aligned,cv2.COLOR_BGR2RGB)
                
                
        depth_image_aligned = depth_image_aligned_org

        depth_to_disparity = rs.disparity_transform(True)
        depth_image_aligned = depth_to_disparity.process(depth_image_aligned)
                
                    
        spatial = rs.spatial_filter()
        spatial.set_option(rs.option.filter_magnitude, 2)
        spatial.set_option(rs.option.filter_smooth_alpha, 0.25)
        spatial.set_option(rs.option.filter_smooth_delta, 20)
        depth_image_aligned = spatial.process(depth_image_aligned)
                    
            
        disparity_to_depth = rs.disparity_transform(False)
        depth_image_aligned = disparity_to_depth.process(depth_image_aligned)
        print('idx',idx)
        cv2.imwrite(os.path.join(save_path,str(idx)+'_rgb.png'),color_image_aligned)
        depth_color_frame = colorizer.colorize(depth_image_aligned)
                
        depth_color_frame = np.asanyarray(depth_color_frame.get_data())
        depth_image_aligned = np.asanyarray(depth_image_aligned.get_data())
        save_image_results(depth_image_aligned / 1000.0, os.path.join(save_path,str(idx)+'_depth_plt.png'))
        cv2.imwrite(os.path.join(save_path,str(idx)+'_depth.png'),depth_image_aligned)
        cv2.imwrite(os.path.join(save_path,str(idx)+'_depth_vis.png'),depth_color_frame)    
        idx += 1
        if idx % 100 == 0:
            print('%d done'%idx)
            
        cv2.imshow("Depth Stream", color_image_aligned)
        key = cv2.waitKey(1)
        if key == 27:
            cv2.destroyAllWindows()
            break
    pipeline.stop()
`
Here the timestamp:
[image]
you can see, the timestamp almost 100ms one frame.

puyiwen, 2023-09-19:
OK, I think I solve the problem, just add
`set_real_time(False) and repeat_playback=False`. I will close the issue.

closed 2023-09-19 by puyiwen

MartyG-RealSense, 2023-09-19:
Thanks very much @puyiwen for the update - it's great to hear that you achieved a solution!