realsenseai/librealsense #12213: Crash during open the D455 ROS bag file

amatskevych, 2023-09-19:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

amatskevych, 2023-09-19:
v2.53.1 works properly

MartyG-RealSense, 2023-09-20:
Hi @amatskevych  Does the error still occur if you launch the 2.53.1 version of the RealSense Viewer first and then close it and launch the 2.54.1 Viewer after that, please?

MartyG-RealSense, 2023-09-20:
I recorded and tested a bag file in the 2.54.1 Viewer in Windows and was not able to replicate your problem.  We do not have Macs to test an RDP connection with in order to replicate your setup, unfortunately.

It is possible to install the RealSense SDK and the Viewer on Mac directly, as RealSense has not officially supported Mac for years but can still work with Mac (up until MacOS Monterey, as Ventura has problems with RealSense at the time of writing this).

amatskevych, 2023-09-20:
I can share my data (BAG file). Because only this data crash the viewer. Other data from  D455 do not crash the viewer.

MartyG-RealSense, 2023-09-20:
Okay, please share the bag and I will test it in my 2.54.1 Viewer in Windows.

amatskevych, 2023-09-20:
Here is the bag file that crashes my viewer: https://drive.google.com/file/d/1uyUy5UYaD4lSE0_bbWsVoFGMFLKAgmZh/view?usp=sharing

MartyG-RealSense, 2023-09-21:
I was able to replicate what you experienced.  Your bag file worked in Viewer 2.53.1 but crashed the 2.54.1 Viewer.  There is no obvious reason why this is occurring for one particular bag file.

amatskevych, 2023-09-21:
Who can debug and find the problem?

MartyG-RealSense, 2023-09-21:
If the problem is only with that specific bag file then there is likely nothing that can be done except to use the 2.53.1 Viewer to play it or try re-recording the bag in 2.54.1 if you have the D455 camera available.

If you download the 2.53.1 version of the Viewer as a standalone .exe (**Intel.RealSense.Viewer.exe**) then you can have the 2.53.1 and 2.54.1 Viewers on the same computer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

amatskevych, 2023-09-21:
I forget to say that BAG file was created on 2.53.1

MartyG-RealSense, 2023-09-21:
I performed a test where I recorded a bag file in 2.53.1 and tried playing it back in 2.54.1 and got the same error.  

The D455 camera had firmware driver 5.14.0.0 installed (which is the recommended firmware for 2.53.1).  I updated the firmware to 5.15.0.2 (the recommended firmware for 2.54.1) and recorded another bag in 2.53.1.  This also could not be played back in 2.54.1.  So the issue does not appear to be related to the firmware driver version that the camera was using at the time of recording the bag.

I recorded a bag in 2.54.1 and had no problem with playing it back in 2.54.1.

So it definitely does seem to be a general problem with playing back bags in 2.54.1 that were recorded in 2.53.1 and not just a problem with your bag recording only.

MartyG-RealSense, 2023-09-27:
Hi @amatskevych  Do you require further assistance with this case, please?  Thanks!

amatskevych, 2023-09-27:
I think the problem is clear. It’s also clear how to get around it. If they check and fix it in the new release, it will be good.

MartyG-RealSense, 2023-09-28:
I have highlighted the issue to my Intel RealSense colleagues.  Thanks again for your report!

MartyG-RealSense, 2023-10-08:
I tested playback of old bag files in the latest Viewer version 2.54.2 and found that the SEH Exception error did not occur for me.

Could you download and run the 2.54.2 version of **Intel.RealSense.Viewer.exe** from the Assets file list and test your bag file in it to see whether the problem still occurs for you?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

MartyG-RealSense, 2023-10-19:
Hi @amatskevych  Do you have an update about this case that you can provide, please?

Have you tried the bag files in version **2.54.2** as I suggested in the comment above?

amatskevych, 2023-10-19:
I tried to open my old files in version 2.54.2  - the same issue
[image]

MartyG-RealSense, 2023-10-19:
As you are continuing to experience this issue in 2.54.2, after discussing the issue with my Intel RealSense colleagues an internal Intel bug report has been created so that the RealSense team can investigate it.

I have added an **Enhancement** label to this case to indicate that it should be kept open whilst the bug report is active, but you do not need to take any further action.  Thanks very much for your report!

MartyG-RealSense, 2024-05-04:
I recorded a bag file in SDK 2.54.2 and tested playback in the current latest 2.55.1 version of the RealSense Viewer.  It played back correctly in 2.55.1.  Would it be possible to confirm whether your old bag files are able to play in 2.55.1 please?

MartyG-RealSense, 2024-08-17:
The bag playback issue was resolved in librealsense SDK version 2.55.1, so I will close this case.  Thanks very much @amatskevych for your earlier reporting of this issue!

closed 2024-08-17 by MartyG-RealSense