realsenseai/librealsense #12214: color and depth frame align too slow on rk3588

westpilgrim63, 2023-09-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11) / Android  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/RK3588  |
| SDK Version                            |  2.0 }                          | 
| Language                            |  {C/java }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
i tried to align color and depth frame using data = align_to.process(data); ,on pc,it's 10 ms,but on rk3588,the same command using about 300 ms ,it's too slow to use in real scene. is there any solution for this?

westpilgrim63, 2023-09-20:
https://github.com/IntelRealSense/librealsense/issues/11758
i found a solution ,using rs2_project_color_pixel_to_depth_pixel to align single pixel

closed 2023-09-20 by westpilgrim63

MartyG-RealSense, 2023-09-20:
Thanks very much @westpilgrim63 for the update.  It's great to hear that you found a solution that works for you!

sun-xm, 2024-06-13:
The problem has not really been solved. Same issue on rk3399. Sdk version 2.50.

MartyG-RealSense, 2024-06-13:
Hi @sun-xm  Depth-color alignment is processed on the CPU of the computer / computing device and not on the camera hardware.  So alignment performance will be partly dependent on the strength of the hardware specification of the RK3399 device.

If programming in C++ language then it is possible to accelerate performance by offloading processing work from the CPU onto the graphics GPU of a device using the RealSense SDK's **GLSL Processing Blocks** feature, described at https://github.com/IntelRealSense/librealsense/pull/3654

There is also a C++ example program called **rs-gl** that demonstrates using GLSL.

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

However, if the computing device has a low-end hardware specification then GLSL may not make a noticeable difference.

sun-xm, 2024-06-18:
Thanks for the tips @MartyG-RealSense. Eventually I've implemented my own version with NEON and OpenMP. Now I can get 30 fps at 1280x720 with around 50% cpu occupation on RK3399.