realsenseai/librealsense #12215: rs2_project_color_pixel_to_depth_pixel inaccurate

westpilgrim63, 2023-09-20:
hi , i used rs2_project_color_pixel_to_depth_pixel to project a point(the red point on the right rgb image) on rgb image to depth image(the red point on the left depth image),but i found the 2d coordinate on depth image in inaccurate ,how can i solve this?
[image]

MartyG-RealSense, 2023-09-20:
Hi @westpilgrim63  The **rs2_project_color_pixel_to_depth_pixel** instruction converts a 2D color pixel to a 3D depth pixel without using depth-color alignment, taking color and depth intrinsic and extrinsic values into account when performing the conversion.  

https://intelrealsense.github.io/librealsense/doxygen/rsutil_8h.html#af04b7b1ba50b1d9e03089d1e5186d2f8

If a color coordinate is projected onto an unaligned depth image as a simple overlay then the position of the dot on the depth image would likely not match the position on the color image because intrinsics and extrinsics between the depth sensor and the color sensor would not be being taken into account.

westpilgrim63, 2023-09-21:
@MartyG-RealSense  is there a way to project several (like 3 or 4) color points to depth points ? align the whole image in a software way is too expensive for rk3588 .align using rs2_project_color_pixel_to_depth_pixel has the above problem
or does rs d435i has some hardware alignning method?

MartyG-RealSense, 2023-09-21:
If you are using C++ then you can use an SDK feature called **GLSL processing blocks** to offload processing of alignment from the computer's CPU onto the GPU.  An example program called **rs-gl** at the link below demonstrates it.

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

In a script that uses alignment, the offloading to GPU would be enabled simply by changing the instruction 'rs2::align' to **rs2::gl::align**

https://github.com/IntelRealSense/librealsense/blob/master/examples/align/rs-align.cpp#L59-L60

GLSL is 'vendor neutral', meaning that it can work with any brand of GPU chip.  However, the effectiveness of the technique will depend on the specification of the computer, and improvement may not be noticable on low-end devices.

westpilgrim63, 2023-09-21:
@MartyG-RealSense thank you anyway!

closed 2023-09-21 by westpilgrim63

Sebtous, 2025-09-21:
If anyone is still having this issue, I found that the `depth_min` parameter of `rs2_project_color_pixel_to_depth_pixel` should be set to a value greater than 0, or it will produce wildly inaccurate results. A depth range of `0.1` (min) to `10.0` (max) seems to work well for me.