realsenseai/librealsense #12216: D435 Frame didn't arrive within 5s

econeale, 2023-09-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.15.02 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.15.0-43-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.54.1                          | 
| Language                            |  N/A                          | 
| Segment			|  others                          | 

### Issue Description
I'm attempting to use the D435 on a Ubuntu 22.04 machine using the Realsense Viewer. When I attempt to connect to the camera I get the "Frame Didn't arrive within 5s" error.

The only thing that seems odd to me is that the device is listed as connected via USB 2.1 in the interface. I tried several different USB cables, just to be sure, as well as all the available ports (including USB C) but that didn't change. Some outputs from `lsusb`: 

```shell
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID c0f4:07b0 USB usb keyboard
Bus 001 Device 009: ID 2575:0001 Weida Hi-Tech Co., Ltd. CoolTouch® System
Bus 001 Device 006: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 008: ID 8086:0b07 Intel Corp. RealSense D435
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```shell
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 4: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 5: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 10: Dev 8, If 4, Class=Video, Driver=uvcvideo, 480M
    |__ Port 10: Dev 8, If 2, Class=Video, Driver=uvcvideo, 480M
    |__ Port 10: Dev 8, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 10: Dev 8, If 3, Class=Video, Driver=uvcvideo, 480M
    |__ Port 10: Dev 8, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 14: Dev 6, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M
```

It looks to me that `xhci_hcd` is identifying the camera as a USB 2.1 device. Does anyone know of a way to modify UDEV rules to force the device to the higher speed? Am I missing something else?

Thanks!

MartyG-RealSense, 2023-09-21:
Hi @econeale  Can you confirm please that all of the USB cables that you are using are USB 3 cables and not USB 2.1 cables?  If a USB 3 device such as a camera is connected to a USB 2.1 cable then the device will be identified as USB 2.1.  This is because USB2 cables lack an additional wire that USB 3 cables have that allows the device to be identified as USB3.

MartyG-RealSense, 2023-09-21:
Thanks very much.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2023-09-27:
It's great to hear that you achieved a solution.  Thanks very much for the update and for the product link!