realsenseai/librealsense #12220: D435i Frames didn't arrived within 5 seconds

andyliao89, 2023-09-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 |  laptop  |
| SDK Version                            |  2.54.1  | 

### Issue Description
I'm attempting to use the D435i on a Ubuntu 20.04 by Realsense Viewer. When I attempt to connect to the camera I get the "Frame Didn't arrive within 5s" error.
[image]
[image]

MartyG-RealSense, 2023-09-25:
Hi @andyliao89  I note that the drop-down menu for selecting a Visual Preset camera configuration option in the side-panel is missing.

[image]

Does the problem still occur if you reset the settings of the RealSense Viewer to their defaults using the instructions at the lnk below, please?

https://support.intelrealsense.com/hc/en-us/community/posts/360052013914/comments/1500000416522

andyliao89, 2023-09-25:
> Hi @andyliao89 I note that the drop-down menu for selecting a Visual Preset camera configuration option in the side-panel is missing.
>
> [image]
>
> Does the problem still occur if you reset the settings [...]

I've tried it and it doesn't work.

andyliao89, 2023-09-25:
> Hi @andyliao89 I note that the drop-down menu for selecting a Visual Preset camera configuration option in the side-panel is missing.
>
> [image]
>
> Does the problem still occur if you reset the settings [...]

After I clicked the advanced mode, the drop-down menu showed up.
[image]

andyliao89, 2023-09-25:
> Thank you. Are you using the official 1 meter USB cable supplied with the camera or a USB cable of your own choice?

I'm using the official ones.

MartyG-RealSense, 2023-09-25:
1.  Has RGB previously functioned correctly and only stopped working today, or has it never worked since the camera was purchased?

2.  Is the RGB problem still present if the computer is shut down completely and then restarted with the power button?

3.  Does unplugging the camera, waiting a couple of seconds and then plugging it back in correct it?

4.  Is RGB still absent if the **rs-capture** example program is launched instead of the Viewer? 

[image]

andyliao89, 2023-09-25:
> 1. Has RGB previously functioned correctly and only stopped working today, or has it never worked since the camera was purchased?
> 2. Is the RGB problem still present if the computer is shut down [...]

1.It has never worked since the camera was purchased.
2.Yes.
3.No.
4.I get nothing if the rs-capture example program is launched.

[image]

[image]

MartyG-RealSense, 2023-09-25:
I will discuss your RGB issue with my Intel RealSense colleagues.  Thanks very much for your patience.

MartyG-RealSense, 2023-09-25:
How long ago was the camera purchased, please?  If it is still within the warranty period then you can contact the retailer that it was purchased from in order to perform an RMA return of the camera to the retailer.

andyliao89, 2023-09-26:
> I will discuss your RGB issue with my Intel RealSense colleagues. Thanks very much for your patience.

Thank you and look forward to your reply.
But now I have a new problem, is there any solution?

[image]
[image]

MartyG-RealSense, 2023-09-26:
I replied above at https://github.com/IntelRealSense/librealsense/issues/12220#issuecomment-1734016033 to ask when your camera was purchased.  Thanks very much.

andyliao89, 2023-09-26:
> I replied above at [#12220 (comment)](https://github.com/IntelRealSense/librealsense/issues/12220#issuecomment-1734016033) to ask when your camera was purchased. Thanks very much.

I bought it in February this year.

MartyG-RealSense, 2023-09-26:
There are sometimes warnings and errors at launch and the camera still works, but a **Right MIPI Error** like the one that you received can indicate that there is a more serious problem involving the hardware inside the camera.

MartyG-RealSense, 2023-10-03:
Hi @andyliao89  Do you require further assistance with this case, please?  Thanks!

andyliao89, 2023-10-03:
Thank you very much !

closed 2023-10-03 by andyliao89