realsenseai/librealsense #12221: High speed cables for data transfer

Nimaro76, 2023-09-25:
Hello,

 Our team recently purchased longer USB-C cables to transfer data from the D435IF camera. However, we've observed that these cables either fail to transfer data or do so at a reduced speed. Could you recommend a source where we can order longer, high-speed cables suitable for data transfer? It's important that the supplier can ship to Europe.

Kind regards,
Nima Roshandel

MartyG-RealSense, 2023-09-25:
Hi @Nimaro76  When using USB cables longer than 2 meters with RealSense cameras, it is important to use high quality USB cables (such as industrial grade ones) that are designed for data transfer.  If the cable length is 5 meters or greater then it is recommendable to use cables with **active repeater** components that boost the signal over distance.  

**Newnex** are a supplier of high quality USB products, including cables with active repeaters, that are validated for use with RealSense cameras.  They are based in the US but ship internationally.

https://www.newnex.com/realsense-3d-camera-connectivity.php

Their products are also available from a selection of Europe-based suppliers.

https://www.newnex.com/where-to-buy.php

MartyG-RealSense, 2023-10-01:
Hi @Nimaro76  Do you require further assistance with this case, please?  Thanks!

Nimaro76, 2023-10-01:
Hello , Thank you for your assistance . The matter is resolved for now.

closed 2023-10-01 by Nimaro76

MartyG-RealSense, 2023-10-01:
That's good news.  Thanks very much for the update!