realsenseai/librealsense #12222: Expose IR stream as a webRTC video stream.

rajrajeswaran, 2023-09-25:
Camera Model: D415
Firmware Version: 05.12.07.100
Operating System & Version: Windows 10 Enterprise 1809 |
SDK Version: 2.38.1
Language: JavaScript

I am building a webRTC sample application and would like to consume IR stream when lighting conditions is low.  This article illustrate how to consume the depth sensor https://github.com/intel/depth-camera-web-demo/blob/master/depth-camera.js .  Is it possible to consume IR stream and any code samples?

MartyG-RealSense, 2023-10-05:
Hi @rajrajeswaran  Do you require further assistance with this case, please?  Thanks!

rajrajeswaran, 2023-10-05:
@MartyG-RealSense I apologize for not closing this question.  I will do it now.

closed 2023-10-05 by rajrajeswaran

MartyG-RealSense, 2023-10-06:
@rajrajeswaran  It's no problem at all.  Thanks very much for the update!