realsenseai/librealsense #12226: "No frames received", No data from camera

lixinghe1999, 2023-09-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (2.54.1) | 
| Operating System & Version |   Win 11  | 
| Platform                                 | PC |
| SDK Version                            | None  | 

### Issue Description
Hello everyone, I am a beginner of realsense, the D435 works well on my friends' laptop, but no my two laptop (same system)
I can clearly see the RGB and Depth module in device manager, however, in 'camera' app from win11, the image is completely dark. In comparison, the integrated RGB on laptop works.

Also, I keep seeing "no frames received" in Realsense viewer. I use the viewer-only version, or install SDK with the viewer, however, but all attempts failed.
According to the below screenshot, it is even impossible to recognize my errors: there is no error in logs.
[image]

I appreciate any help! Thank you in advance.

MartyG-RealSense, 2023-09-27:
Hi @lixinghe1999  Please first check in the Windows camera permissions that permission to access the camera hardware and permission for applications to access the camera are both set to On, using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

lixinghe1999, 2023-09-27:
> Hi @lixinghe1999 Please first check in the Windows camera permissions that permission to access the camera hardware and permission for applications to access the camera are both set to On, using the instructions at [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)

Thank you for the rapid reply. Unfortunately, I found all the setting is on. I also install the official driver (https://www.intel.com/content/www/us/en/download/19236/intel-realsense-d400-series-universal-windows-platform-uwp-driver-for-windows-10.html?), but there is no luck. 

Actually the viewer can recognize the usb as RealSense but there is no data (either "no frames received" or complete dark for camera). Can you point out some potential solutions?

lixinghe1999, 2023-09-27:
A good news is that I find the depth quality tool gives me some different things.
I can get some data now but it will receive "no frames received" soon and the image will not be updated.
[image]

I have two questions:
1. Why viewer and depth quality tool performs differently?
2. Why the depth quality tool can also receive several seconds data and go to "no frames received" again?

MartyG-RealSense, 2023-09-27:
1.  The Viewer and Depth Quality Tools have common elements to their design and programming and so tend to behave similarly.  They have different purposes though, with the Viewer focused on displaying camera data visually and modifying it, and the Depth Quality Tool analyzing depth images for their quality and accuracy.

2.  Does performance improve if you disable the **GLSL** option in the settings interface of the Viewer and Depth Quality tools using the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

lixinghe1999, 2023-09-27:
> 1. The Viewer and Depth Quality Tools have common elements to their design and programming and so tend to behave similarly.  They have different purposes though, with the Viewer focused on displaying camera data visually and modifying it, and the Depth Quality Tool analyzing depth images for their quality and accuracy.
> 2. Does performance improve if you disable the **GLSL** option in the settings interface of the Viewer and Depth Quality tools using the instructions at [No 3D point cloud in RealSense Viewer #8110 (comment)](https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023)

Thank you for your reply!
with the GLSL disabled I still can't access data via viewer, meanwhile, the depth quality get "No frames received" as well after few seconds. Do you have any ideas?

MartyG-RealSense, 2023-09-27:
If you are using modern Windows 11 laptops then I would expect the USB ports to be USB 3.2 speed, but the Viewer and Depth Quality tools are detecting the camera as the slower USB 2.1.  If both of your laptops are the same model then it could suggest that the brand of USB port controller used by that computer model does not work well with RealSense cameras.

You could test this theory by attaching a USB hub to the laptop and plugging the camera into the hub instead of directly into the built-in USB port of the laptop, as the ports of USB hubs use their own USB controller separate from the brand of controller used by the computer.

lixinghe1999, 2023-09-28:
I made it by a weird combinations:
first connect RealSense by USB-C To USB-A and insert a USB3.0 PORT in a usb hub. Then the USB hub is inserted to the PC. Right now the data seems good but the USB is still 2.1 (seems have no disadvantage at least now). I will move to other functions of RealSense and will post the newest updates.
[image]

MartyG-RealSense, 2023-09-28:
It's excellent to hear that you have made progress.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2023-10-12:
You are very welcome.  I will close the case for you.  Thanks for the update!

closed 2023-10-12 by MartyG-RealSense