realsenseai/librealsense #12227: Stream live video over ethernet

GirikrishnaT, 2023-09-27:
[image]

The figure above is a proposed data path of the video stream. 
Camera Model D435. 
We are using Realsense SDK to read the RGB video stream from the Camera. The stream has to be sent via ethernet to a edge gateway and subsequently to the Cloud. We plan to use Kinesis platform from AWS to handle the video stream on the cloud.
On the device, the programming language would be C++, OS is Yocto Linux. 

Do you have samples where the video stream read from USB is sent across to the Edge gateway via ethernet.

We may want to downscale the incoming video stream (from the camera) not to choke the network and reduce the cost on the cloud.
Any samples related to this would also help.

MartyG-RealSense, 2023-09-27:
Hi @GirikrishnaT  I located a research paper that describes using a RealSense D435i camera with a Dell Edge Gateway 5000.  Clicking on the link below will download a PDF of the paper to your computer.  Page 4 of the paper has details of using the camera via USB with the Edge Gateway, "which acted as a server for the operator to connect to and stream the video and depth stream back to the workstation".

https://researchrepository.ul.ie/articles/conference_contribution/Enabling_communications_for_industry_4_0_Private_5G_in_smart_manufacturing/23945901/1/files/41994243.pdf

Another paper that uses RealSense D435 and an edge gateway can be downloaded as a PDF from here:

https://ec.europa.eu/research/participants/documents/downloadPublic?documentIds=080166e5bd4c3ead&appId=PPGMS

MartyG-RealSense, 2023-10-03:
Hi @GirikrishnaT  Do you require further assistance with this case, please?  Thanks!

GirikrishnaT, 2023-10-04:
@MartyG-RealSense , Thanks for pointing out to the research papers. These papers however are conceptual and what i was also looking for is sample C++ code.   Are there samples that i could use which can read the stream from the Camera and pass this on another server (Edge) via Ethernet, say using Sockets

MartyG-RealSense, 2023-10-04:
There is not a C++ ethernet code example that meets your description, unfortunately.  There is a Python-based RealSense ethernet tool called **EtherSense** that uses sockets.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

closed 2023-10-17 by MartyG-RealSense