realsenseai/librealsense #12235: I can't call the D435i camera through matlab2021b, is there any sample code?

hanbaoguai, 2023-09-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  windows 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/windows 10   |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  matlab | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

hanbaoguai, 2023-09-29:
I want to get the image information in real time through MATLAB. how to write the code? thank you

hanbaoguai, 2023-09-29:
[image]
Error in depth_example (line 3)
pipe = realsense.pipeline();

hanbaoguai, 2023-09-29:
[image]

this error appears when I run depth_example.m, do you know how to fix it? Thank you

MartyG-RealSense, 2023-09-29:
Hi @hanbaoguai  There have been a couple of past cases of this error when using the MATLAB wrapper program **depth_example.m**.  The MATLAB wrapper is not fully compatible with modern MATLAB versions such as 2020b and 2021b and errors can result from using these versions.  The error does not occur when using MATLAB 2017b.  There are no plans to update the wrapper for modern versions, unfortunately.

A RealSense user who experienced this error created their own MATLAB depth script at https://github.com/IntelRealSense/librealsense/issues/9773#issuecomment-925553900 that worked for them.

hanbaoguai, 2023-09-30:
I have replaced matlab2017b，
[image]
This problem still persists, what could be the cause?thank you

hanbaoguai, 2023-09-30:
I have added the +realsense folder to the MATLAB path，The camera can also be used normally in the SDK，However, the above error still occurs when using matlab.

hanbaoguai, 2023-09-30:
[image]

MartyG-RealSense, 2023-09-30:
Is your path **+realsense/depth_example.m** like in the image below, please?

[image]

Are any of the other MATLAB wrapper examples able to work?

advanced_mode_example.m
capture_example.m
depth_view_example.m
pointcloud_example.m
rosbag_example.m

hanbaoguai, 2023-09-30:
All examples don't work. Is it possible that this problem is related to the sdk? My current version is 2.54.1. I'll try downloading an earlier version to see if it works.

MartyG-RealSense, 2023-09-30:
If you downgrade to an earlier version then the camera firmware driver installed in your camera should be changed to match a particular SDK version.

2.54.1: firmware 5.15.0.2
2.53.1: firmware 5.14.0.0
2.50.0 and 2.51.1: firmware 5.13.0.50

hanbaoguai, 2023-10-01:
I have solved this problem and I will explain my approach below.
1. the problem is not related to the SDK version
2. Both MATLAB2017b and MATLAB2021b can be used normally without reporting errors
My method is as follows: first of all, you need to set the whole Intel RealSense SDK 2.0 folder into maltab path, when you add it, choose include subfolder, and set all of them into matlab path. if you have the same error as me, please run capture_m first, and then you can call the camera through other codes.

hanbaoguai, 2023-10-01:
I have solved this problem and I will explain my approach below.

1.the problem is not related to the SDK version
2.Both MATLAB2017b and MATLAB2021b can be used normally without reporting errors
My method is as follows: first of all, you need to set the whole Intel RealSense SDK 2.0 folder into maltab path, when you add it, choose include subfolder, and set all of them into matlab path. if you have the same error as me, please run capture_example.m first, and then you can call the camera through other codes.

MartyG-RealSense, 2023-10-01:
It's excellent news that you achieved a solution.  Thanks so much for sharing what worked for you!

closed 2023-10-08 by MartyG-RealSense