realsenseai/librealsense #12236: What happens when ROI is selected ?

Biopticon, 2023-09-29:
I want to display and process only the selected ROI. This way I display only the area of interest and process a lot less amount of data. But nothing changes visually after I do that.  Am I missing something?

MartyG-RealSense, 2023-09-30:
Hi @Biopticon  In the RealSense SDK, the Region of Interest (ROI) does not crop the image.  It instead attempts to keep the intensity of the pixels in the ROI area at a pre-defined 'setpoint' value.

It is possible to crop a depth image horizontally and vertically, though it is complicated to do so.  https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157 provides an example provided by a RealSense user of how they approached generating cropping.

An alternative to a ROI for you may be to define a post-processing **Decimation Filter** that reduces the complexity of the depth image by downsampling its resolution.

https://dev.intelrealsense.com/docs/post-processing-filters#decimation-filter

MartyG-RealSense, 2023-10-08:
Hi @Biopticon  Do you require further assistance with this case, please?  Thanks!

Biopticon, 2023-10-08:
Hello Marty,

Thank you for your prompt response and useful information.

I want to obtain morphological information with submillimeter resolution using any scanner from the D series.  The D405 is desirable since it has the shortest scan distance, I need to scan small objects with about an inch cubed.  Its depth accuracy is mentioned to be "+/-2% at 50cm" in the specs which is not as claimed to be "With defect detection accuracy as good as 0.1mm" in the "Industrial Solutions that work" section of the website. So, what are the real measurement accuracies in-depth and, in the x, and y dimensions for D405?

Even though the higher models such as D435 have Structured Light capabilities with better-expected resolution than a stereo system like D405, they are intended to scan at a lot larger distances.  But then, is it possible to achieve a submillimeter depth accuracy with, say, using D435 at 28cm, minimum depth distance (Min‑Z) at max resolution?

I also need high speed for the post-processing of the Point Cloud data, in the region of 1s. I hoped I could achieve this by restricting the scan area to only 4x4cm2 but as you mentioned we cannot define a dynamic ROI to acquire point cloud data.

So, the real question is, if I use the SDK provided, can I define ROI, obtain data, produce a point cloud of the scanned region, in only ~<1s, and access this data to CPU to do post-processing with a program not being limited by the functions supplied by the GUI?

Can we communicate with you privately?

Sincerely,

Cem

MartyG-RealSense, 2023-10-09:
D435 is a stereo depth camera like D405.  The only RealSense camera models that used technology similar to structured light were the F200 and SR300 / SR305, which are obsolete and retired.

D405 is specially designed for high accuracy, high quality images at very close range.  Accuracy will improve as the camera is moved closer to a surface.  A discussion at the link below talks about D405 submillimeter accuracy.

https://support.intelrealsense.com/hc/en-us/community/posts/20608351547795-Need-information-related-to-accuracy-of-ReaslSense-cameras

For some surfaces the results from D405 will be better than D435 could provide at that range.  However, because the D405 is not equipped with a dot pattern projector, some types of surfaces such as plain smooth ones with low texture or no texture will be more difficult to read depth information from than with D435.

D405 is the only RealSense model that is officially described as being capable of submillimeter accuracy.

D435 supports an 848x100 resolution 'high speed capture' mode that can provide 300 FPS but only has a processing burden equal to 90 FPS.  It does this by restricting the viewpoint to a 100-high vertical strip at the center of the image.  More information about this can be found in Intel's white-paper guide on the subject at the link below.

https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

If you wrote a program in C++ language then you could also take advantage of the SDK's support for GLSL Processing Blocks, which can make use of a graphics GPU to accelerate the processing of functions such as pointcloud and alignment.  GLSL is 'vendor neutral', meaning it can work with any GPU brand and not just a specific one such as Nvidia.

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

If you wish to discuss a confidential matter that you cannot write about publicly then you can email me.

martyx.grover@intel.com

MartyG-RealSense, 2023-10-19:
Hi @Biopticon  As your case is being handled by Intel outside of this support forum, do you wish to keep this GitHub case open for a further time or close it, please?  Thanks!

MartyG-RealSense, 2023-10-25:
Case closed due to no further comments received.

Biopticon, 2023-10-25:
Hello Marty,

      We are anxiously trying to understand whether we could use D405 as our 3D scanner. As you suggested we have started talking to RealSense apps developer companies about this. But we are still not sure about the accuracy of measurements by this scanner, i.e. if I scan an object, say 10mm in length,  width, and height, what D405 will produce on x, y, and z dimensions, 10.0, 10.5 or 11.0mm at 7-10cm working distance? Intel says on the website that "defect detection accuracy as good as 0.1mm" while depth accuracy in the specs is reported as <2% at 50 cm, and also not at 7 cm minimum distance which is more relevant for this model. We would like to get a proper understanding of this accuracy issue.

       As importantly, we are also interested in the RMS noise performance of the D405 when imaging similarly textured surfaces from a distance of 70-100mm. Our application requires measuring deformations on these surfaces about 1mm deep and perhaps as small as 4mm^2 in area. As a target for noise performance, an RMS pixel noise of ~0.1mm would be required on the final depth map generated. You mentioned earlier that you did a study on RMS noise at 70cm. Could you also please tell us what the RMS noise is at 100mm? This is an important parameter for us.

      Our application is relatively simple: single scan, stationary and small FOV but we need sub-millimeter accuracy on all 3 dimensions. It seems that PC creation and transfer time is short enough not to worry about.

      No salesperson has contacted us so far.

      Looking forward to hearing from you on the issues we raised above.

      Sincerely,

Cem

MartyG-RealSense, 2023-10-26:
I will follow up with Intel about your questions above and the salesperson contact.  I will also re-open this case.

Biopticon, 2023-10-26:
Thank you very much, Marty. We are a small company, but we are contributing a lot to cancer research with our Tumor Scanner.   Any help will help to advance our technology and lower our costs so that more scientists can use our system.

Sincerely,

Cem

jnsweets, 2023-10-27:
Hi Cem,

I will try to address most, if not all, of your issues and questions. This may result in a longer dialogue to fully resolve everything but I'll start by responding to the specific items in the last comment above.

_if I scan an object, say 10mm in length,  width, and height, what D405 will produce on x, y, and z dimensions, 10.0, 10.5 or 11.0mm at 7-10cm working distance?_
The answer to this depends on more details of the setup and operating conditions. 

- As mentioned earlier, the D405 does not include an integrated light or texture source and so the object of interest needs to be sufficiently illuminated and include some "natural" texture in order to obtain the best results. 
- You also mention this will be a "single scan". I interpret this to be static capture of the object from a fixed viewpoint. If the camera or object is actually scanned (moved), the results will be different (generally better). 
- The timescale over which the measurement occurs is also important - generally, the longer, the better. 
- Finally, the detailed shape of the object is also relevant. For example, the accuracy of measuring the outer dimensions of the object may be different from the ability to measure the size and shape of a surface deformation.

With the above caveats, and making a few simplifying assumptions, I can give some approximate answers. 
The accuracy specs provided are for absolute distance from camera to object. While this affects the accuracy of the object measurements, the actual dimensional accuracy should be better than this in general but are different for each of the x,y, and z dimensions. 
Z (depth): At a distance of ~10cm, the accuracy that an object's depth (e.g., height) can be measured depends on the height but should be well within 0.1mm for objects up to ~10mm with accuracy scaling with height. So, the accuracy of a ~1mm height change would be ~0.01mm. This is in the absence of any noise and based on the systematic depth error from the camera. RMS noise adds to these values and is determined by a variety of factors (e.g., texture, lighting, degree of filtering used). A rough number for moderate texture (e.g., comparable to wood grain), good lighting, and minimal filtering would be ~0.3% or ~0.3mm at a distance of 10cm. This could be improved with additional processing.
X, Y (lateral dimensions): This is really a question of spatial resolution. It is affected by some of the same factors as depth but is largely determined by the camera's resolution. In general, the spatial resolution of the depth image is several pixels (e.g., an depth edge/step spans ~4-8 pixels. If the corresponding RGB image is used to define the edge, this can be reduced to 1-2 pixels. For context, a D405 pixel at 100mm and HD resolution is ~0.15mm.

I've attached a few sample images of a wooden block with carved features that are on the scale of your application (~1mm deep, 10mm wide). The images were captured at a distance of ~100mm. This is to give you a sense of the what the camera is capable of producing. My conclusion is that it is feasible to meet your requirements but will require near optimal conditions and some additional processing.

If this looks sufficiently promising to you, my suggestion would be to obtain some cameras and test out the capabilities on actual test objects of interest.

Of course, feel free to follow-up with any questions or comments.

John
[D405_Depth Image Examples_231027.pdf](https://github.com/IntelRealSense/librealsense/files/13193528/D405_Depth.Image.Examples_231027.pdf)

Biopticon, 2023-10-31:
Hello John,

      Thank you very much for this useful information.  Is our correspondence private between us, or is it shared in GitHub? I will respond to you in detail if it is only for you. I already signed Intel's NDA.

           Sincerely,

Cem

I. Cem Girit, PhD

[cid:ebebd5f4-59cb-4271-8eae-0a4a863ebd9a]
Biopticon Corporation
33 State Road, Suite F
Princeton, NJ 08540-1304
Tel: (609)-853-0231
Email: ***@***.******@***.***>
Website: www.biopticon.com<https://urldefense.proofpoint.com/v2/url?u=http-3A__www.biopticon.com_&d=DwMFAg&c=CZZujK3G2KuXGEKnzq-Hrg&r=ckqPL-NvwfzepOqM90Z3Ie-yByj7xS1_kk2JaSjFdp8&m=USdbKWLLAqHvmjycnGvbwnLAmCuMjMAwZ3ufYrlKKVM&s=J7K3OtUzMmMaM0mX9apPXxMfYDfoAzv8uP3x15wOXA8&e=>

This email message and any attachments are confidential and may be privileged. If you are not the intended recipient, please notify Biopticon Corporation immediately -- by replying to this message or by sending an email to ***@***.******@***.***> -- and destroy all copies of this message and any attachments. Thank you.

jnsweets, 2023-10-31:
Hi Cem,

Github is a public forum, but I see this issue is already linked to a Zendesk ticket. You can reply in private on that ticket:

Ticket #21553
https://intel.zendesk.com/agent/tickets/21553

We can continue correspondence via Zendesk.

Thanks,
John

Biopticon, 2023-11-01:
Hello John,

      That link takes me to a nonexistent site. See below. Shall I instead use the https://realsensesupport.intel.com/hc/en-us/requests/new site with ticket #21553?

Cem

[cid:dfc4cba9-9550-4fe1-835a-e137f210f885]

https://realsensesupport.intel.com/hc/en-us/requests/new

https://intel.zendesk.com/agent/tickets/21553

MartyG-RealSense, 2023-11-01:
Hi Cem, I tested the link and it worked correctly for me.  It may be an issue with your login details that is causing the access issue.

Biopticon, 2023-11-01:
Hello Marty,

      Thank you for helping with this. After trying another browser, I was able to get to the following site (see below). I will then go to Submit a Request then mention ticket#21553?

Cem

[cid:99bd9fbf-5fd8-4c05-8f5d-93a3981958f4]

jnsweets, 2023-11-01:
Cem,

The easiest resolution might be to just submit a new ticket (you can refer to #21553). We should be able to link them in the Zendesk system.

Let us know if you have more issues.

John

MartyG-RealSense, 2023-11-09:
Hi @Biopticon  As you have submitted a private Zendesk ticket, can this public GitHub issue now be closed please?  Thanks!

closed 2023-11-21 by MartyG-RealSense