realsenseai/librealsense #12238: How to calculate distance from processed depth data

Lalala-ysl, 2023-09-30:
hello
I used python to read the data from the SR305 depth camera as follows:
aligned_frames = align.process(frames)
# Get aligned frames
aligned_depth_frame = aligned_frames.get_depth_frame()
depth_image = np.asanyarray(aligned_depth_frame.get_data())
I used my own algorithm to complete the missing depth data in depth_image. Then I wanted to restore the completed depth_image to aligned_depth_frame for calculating the distance. But I looked for a lot of relevant information and didn't find that anyone had solved it. If you know trouble give me some advice, thank you very much for your help!
[image]

MartyG-RealSense, 2023-09-30:
Hi @Lalala-ysl  Does it make a difference if you insert copies of the `aligned_depth_frame` and `depth_image` lines above fill_deep_data to update the contents of aligned_depth_frame and depth_image after you have performed the fill operation?

```
aligned_depth_frame = aligned_frames.get_depth_frame()
depth_image = np.asanyarray(aligned_depth_frame.get_data())

fill_deep_data = deepFill(item, depth_image)
```

Lalala-ysl, 2023-10-08:
> Hi @Lalala-ysl Does it make a difference if you insert copies of the and lines above fill_deep_data to update the contents of aligned_depth_frame and depth_image after you have performed the fill operation?`aligned_depth_frame``depth_image`
> 
> ```
> aligned_depth_frame = aligned_frames.get_depth_frame()
> depth_image = np.asanyarray(aligned_depth_frame.get_data())
> 
> fill_deep_data = deepFill(item, depth_image)
> ```

First of all, thank you very much for your reply. What I want to achieve is to calculate the distance between two points through the depth information. However, many depth data in the depth information captured by the SR305 camera are missing, which makes it impossible for me to measure the target I want, so I complete the missing depth data myself.
This is the measurement of the distance between two points using the function provided by the official:

> ```
> def calculate_distance(x, y, ix, iy, color_intrin, depth_frame):
>     udist = depth_frame.get_distance(ix, iy)
>     vdist = depth_frame.get_distance(x, y)
>     # print udist,vdist
> 
>     point1 = rs.rs2_deproject_pixel_to_point(color_intrin, [ix, iy], udist)
>     point2 = rs.rs2_deproject_pixel_to_point(color_intrin, [x, y], vdist)
>     # print str(point1)+str(point2)
> 
>     dist = math.sqrt(
>         math.pow(point1[0] - point2[0], 2) + math.pow(point1[1] - point2[1], 2) + math.pow(
>             point1[2] - point2[2], 2))
>     # print 'distance: '+ str(dist)
>     return dist
>frames = pipeline.wait_for_frames()
>aligned_frames = align.process(frames)
>
># Get aligned frames
>aligned_depth_frame = aligned_frames.get_depth_frame()  # aligned_depth_frame is a 640x480 depth image
>color_frame = aligned_frames.get_color_frame()
>color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
>distance = calculate_distance(x_ysl, y_ysl, x_ysl1, y_ysl1, color_intrin, aligned_depth_frame)              
> ```

But now I complete the missing depth data, the code：

> ```
> def calculate_distance(x, y, ix, iy, color_intrin, depth_frame):
>     udist = depth_frame.get_distance(ix, iy)
>     vdist = depth_frame.get_distance(x, y)
>     # print udist,vdist
> 
>     point1 = rs.rs2_deproject_pixel_to_point(color_intrin, [ix, iy], udist)
>     point2 = rs.rs2_deproject_pixel_to_point(color_intrin, [x, y], vdist)
>     # print str(point1)+str(point2)
> 
>     dist = math.sqrt(
>         math.pow(point1[0] - point2[0], 2) + math.pow(point1[1] - point2[1], 2) + math.pow(
>             point1[2] - point2[2], 2))
>     # print 'distance: '+ str(dist)
>     return dist
>frames = pipeline.wait_for_frames()
>aligned_frames = align.process(frames)
>
># Get aligned frames
>aligned_depth_frame = aligned_frames.get_depth_frame()  # aligned_depth_frame is a 640x480 depth image
>depth_image = np.asanyarray(aligned_depth_frame.get_data())
>fill_deep_data = deepFill(depth_image)  #This is my own implementation of the depth information completion function    
> ```

 **I can't restore fill_deep_data to aligned_depth_frame at the moment, so I can't use any of the following code**

> ```
>color_frame = aligned_frames.get_color_frame()
>color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
>distance = calculate_distance(x_ysl, y_ysl, x_ysl1, y_ysl1, color_intrin, aligned_depth_frame)            
> ```
**fill_deep_data** is of type **np.array**, and **calculate_distance** function needs the same type of data as **aligned_depth_frame**. How do I update the depth data in **aligned_depth_frame** to **fill_deep_data** so that I can calculate the depth information between those two points after completion using the **calculate_distance** function

MartyG-RealSense, 2023-10-08:
It looks as though you are using the **measure_new.py** point to point measuring script as the basis of your code.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

If you are then you are probably using `align_to = rs.stream.color` in your script, which aligns the depth image to the color image.  Does the aligned image improve if you align **color to depth** instead?

`align_to = rs.stream.depth`

Color to depth alignment is used in the original C++ version of the measure example.

https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp#L133

********

You could also make sure that the **Laser Power** value is maximized in order to see whether it helps to fill in missing detail on the depth image.

`depth_sensor.set_option(rs.option.laser_power, range.max)`

Lalala-ysl, 2023-10-09:
> It looks as though you are using the **measure_new.py** point to point measuring script as the basis of your code.
> 
> https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py
> 
> If you are then you are probably using in your script, which aligns the depth image to the color image. Does the aligned image improve if you align **color to depth** instead?`align_to = rs.stream.color`
> 
> `align_to = rs.stream.depth`
> 
> Color to depth alignment is used in the original C++ version of the measure example.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp#L133
> 
> You could also make sure that the **Laser Power** value is maximized in order to see whether it helps to fill in missing detail on the depth image.
> 
> `depth_sensor.set_option(rs.option.laser_power, range.max)`

Thanks for your patient answer, I have found a new way to calculate this distance.

MartyG-RealSense, 2023-10-09:
You are very welcome.  I'm very pleased to hear that you found a solution!

closed 2023-10-14 by MartyG-RealSense