realsenseai/librealsense #12246: How to color a point cloud？

pgswag, 2023-10-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  MATLAB | 

### Issue Description
My shooting scene is shown below：
[image]

The result of running 'pointcloud_example.m' in MATLAB is shown below：
[image]

How do I color it so that it looks like the image below：
[image]

The figure of below is the code in pointcloud_example. The code in the red box appears to be coloring the pointcloud, but when I uncomment it and run it, I get the same result as before.
[image]

MartyG-RealSense, 2023-10-02:
Hi @pgswag  Adding the **colorizer** code from lines 5 and 26 of the **depth_example.m** example to the appropriate places in the **pointcloud_example.m** script may work.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/depth_example.m

**Line 5**

```
pipe = realsense.pipeline();
% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();
```

**Line 26**

```
depth = fs.get_depth_frame();
% Colorize depth frame
color = colorizer.colorize(depth);
```

pgswag, 2023-10-02:
Here's the code I changed, but it doesn't seem to be working
```
function pointcloud_example()
    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    % Make Colorizer object to prettify depth output
    colorizer = realsense.colorizer();
    pointcloud = realsense.pointcloud();

    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();

    figure('visible','on');  hold on;
    figure('units','normalized','outerposition',[0 0 1 1])

    % Main loop
    for i = 1:10000

        % Obtain frames from a streaming device
        fs = pipe.wait_for_frames();
        
        % Select depth frame
        depth = fs.get_depth_frame();
        color = fs.get_color_frame();
        % Colorize depth frame
        color = colorizer.colorize(depth);
        % Produce pointcloud
        if (depth.logical() && color.logical())

            pointcloud.map_to(color);
            points = pointcloud.calculate(depth);
            
            % Adjust frame CS to matlab CS
            vertices = points.get_vertices();
            X = vertices(:,1,1);
            Y = vertices(:,2,1);
            Z = vertices(:,3,1);

            plot3(X,Z,-Y,'.');
            grid on
            hold off;
            view([45 30]);

            xlim([-0.5 0.5])
            ylim([0.3 1])
            zlim([-0.5 0.5])

            xlabel('X');
            ylabel('Z');
            zlabel('Y');
            
            pause(0.01);
        end
        pcshow(vertices); 
    end

     % Stop streaming
    pipe.stop();
    
end
```

MartyG-RealSense, 2023-10-02:
I wonder if this line is causing confusion for the program:

 color = colorizer.colorize(depth);

Because you are already using the 'color' variable name for the RGB color stream.

Does it make a difference if you change the name of the colorizer variable to 'colorized'

` colorized = colorizer.colorize(depth);`

pgswag, 2023-10-02:
Thank you very much for your patient answer, but it still hasn't worked.

MartyG-RealSense, 2023-10-02:
You are very welcome.  It's excellent to hear that the script resolved your problem and I am pleased that I was able to help.  :)

closed 2023-10-03 by pgswag