realsenseai/librealsense #12254: D457 RGB Stream Flickering Rapidly

robotom, 2023-10-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457| 
| OS                   | Ubuntu 20 | 
| Platform			|  NVIDIA Orin                          | 

#### Issue Description:

Hello, we're using the RGB imager on the new D457 and we've noticed a flickering. Perhaps, auto exposure related. It makes the images somewhat unusable for training our models. 

We've added some [starwave ND filters](https://starwavelabs.com/) to remove the purple issue that the Realsense has and for now, the problem appears not to be there. But, this could also just be luck of the particular boot, the sun, time of day, etc. With the addition of the filters, we are now seeing a bit of motion blur. I will post back after some more testing.

- Has anyone seen this flickering before? 
- Are there any known fixes?
- Thanks in advance!

#### Video (unfiltered): 

[image]

MartyG-RealSense, 2023-10-04:
Hi @robotom  I have not seen that particular alternating effect before.  Could you test whether having Auto Exposure enabled and disabling the RGB option **Auto Exposure Priority** rectifies the problem, please?

You can test disabling of Auto Exposure Priority in the **RealSense Viewer** tool by going to the **RGB > Controls** section of the options side-panel.

[image]

robotom, 2023-10-09:
@MartyG-RealSense Thanks for the reply! I have not been able to replicate the problem using the auto exposure setting on/off. Sometimes it happens. Sometimes it doesn't. Via the ros launch file I have an exposure region of interest set as well. That might have helped.

MartyG-RealSense, 2023-10-09:
I note that near the bottom of the image there is a horizontal line across the image that appears and then disappears.  In past cases of that effect it has suggested that a small part of the image is out of sync with the updating of the rest of the image, though in your case the effect is not serious as the affected section at the bottom is almost exactly in time with the rest of the image above it and so the horizontal line is the only negative effect.

MartyG-RealSense, 2023-10-15:
hi @robotom  Do you require further assistance with this case, please?  Thanks!

robotom, 2023-10-16:
@MartyG-RealSense the line is just an artifact of a video player that this animated image was screenshotted from I think. I haven't been able to replicate the problem since applying the filters and since changing the backlight parameter in the launch file. I guess for now it would be okay to close, can it be reopened later if the problem arises again?  Thanks!

MartyG-RealSense, 2023-10-16:
Yes, it is fine to close this case using the button under the comment writing window and re-open it at a future time if you need to do so.

closed 2023-10-16 by robotom