realsenseai/librealsense #12256: realsense issue

peter8787-afk, 2023-10-04:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2023-10-04:
Hi @peter8787-afk  This message indicates that the graphics driver on your Windows computer may not have **OpenGL** support.  I recommend updating your computer's graphics driver to see whether it resolves the issue.

MartyG-RealSense, 2023-10-19:
Case closed due to no further comments received.

closed 2023-10-19 by MartyG-RealSense

hwdworld, 2024-06-21:
You can download OpenGL compatibility package through Microsoft Store to solve this problem

MartyG-RealSense, 2024-06-21:
Thanks very much @hwdworld for sharing your advice!